Lora-net/LoRaMac-node #1015: LinkAdrReq re-enabling old channels

clemehaute, 2020-12-21:
Hello,
During a EU868 certification test using LCTT, a behavior which may be inconsistent has been highlighted.
In this test, additional channels are enabled via the CFList in a first Join Session and a little bit later a new Join Session without additional channels is created. And the reception of a LinkAdrReq with a ChMaskCntl value set to 6, re-enables the channels created in the first Join Session.
It could be prevented by clearing the frequency field of non-mandatory channels in the `RegionEU868InitDefaults` function. I didn't look but this may also impact other Regions.

mluis1, 2021-03-03:
Could you please provide more information?

- Which version or commit  hash of this project are you using?
- Is the NVM context management enabled?

While running the LCTT certification tests we haven't observed the reported issue and all tests are running successfully.
Please refer to [Releases pre-certification results](https://github.com/Lora-net/LoRaMac-node/wiki/releases-pre-certification-results) wiki page for further information on the pre-certification status.

clemehaute, 2021-03-03:
Hi Miguel,
I am using the ST layer of the LoRaWAN based on an old commit (develop/v4.4.4 `ba17382bd5109513937afad07f068a781a503ef6`) without the NVM context management enabled.
If I remember correctly, what I saw is that new channels can be added in a first Join session, populating the `Channels` member of the `RegionEU868NvmCtx_t` (called `RegionNvmDataGroup2_t` in the newest stack version). And when a new join is requested the `ChannelMask` is reset to only use the default channels but `Channels` is left untouched, and a LinkAdrReq command with ChMaskCntl value set to 6 would re-enable the channels created in the first Join Session since their are still present but only disable by the `ChannelMask`. But maybe it is only relevant to the ST layer or a mistake on my side.
Regards

mluis1, 2021-03-03:
Thanks for the provided information.

In version v4.4.5 we have refactored the way the channels are initialized.
Please take a look at following commit https://github.com/Lora-net/LoRaMac-node/commit/e7cb85494143ef7138c2660d9055a893b4bdb135.

Please verify that ST port includes this fix as it may solve the issue you are observing.

mluis1, 2021-03-03:
Something strange to be noted is that we successfully run LCTT tests for EU868 when we have made the 4.4.4 release.

Maybe the LCTT tests have been updated in the meantime and now they catch the reported issue.

clemehaute, 2021-03-18:
Hi Miguel,
Yes, the ST layer implements the commit https://github.com/Lora-net/LoRaMac-node/commit/e7cb85494143ef7138c2660d9055a893b4bdb135, they must have cherry-picked it.
I confirm that what I see is that the test TP_A_EU868_ED_MAC_BV_008 (Technology package 2.6.0) reenables a non-default channel passed in the CFList during the test TP_A_EU868_ED_MAC_BV_001_A, after a new Join and a LinkAdrReq with a ChMaskCntl to 6.
I will force all non-default channels frequency to 0 during a new Join (INIT_TYPE_RESET_TO_DEFAULT_CHANNELS operation) to pass the tests.
If you are sure that the issue is not present in the new versions of the stack, you can close this ticket.
Thanks for your help.
Regards,
Christopher

mluis1, 2021-03-18:
OK, Thanks for the feedback. I close the issue then.

closed 2021-03-18 by mluis1