Lora-net/LoRaMac-node #1017: LoRaWAN Certification Issues

AsteelDev, 2020-12-22:
Hello, 

We are certifying a LoRaWAN Product and we get a few failed test cases.

The system architecture is as follows:
  - LoRa Stack Release 4.4.5.
  - BlueNRG-2 
  - sx1262 (TCXO)
  - Systick instead of RTC

Could you help usplease to resolve these issues.
The fact that we don't RTC can be the cause?

I report below themain  error message we got for each test. For more details, I attached the Test Control Layer [TCL] logs 

**TP_A_EU868_ED_MAC_BV_015_A : RX1 Receive window test (Part 1)**     
  -  Test packet error rate for SF12BW125 using RX1
[TP_A_EU868_ED_MAC_BV_015_A.zip](https://github.com/Lora-net/LoRaMac-node/files/5730676/TP_A_EU868_ED_MAC_BV_015_A.zip)

**TP_A_EU868_ED_MAC_BV_001_A : Over The Air activation**
  - Default channels (868.1, 868.3 and 868.5 MHz) have   not been used   
  - Activation (OTAA) process has not finished   successfully   
  - Error in Trigger Join Request communication
  
[TP_A_EU868_ED_MAC_BV_001_A.zip](https://github.com/Lora-net/LoRaMac-node/files/5730681/TP_A_EU868_ED_MAC_BV_001_A.zip)

`

47 | Start Test Case: | mod_LORA_A_EU868.TC_A_EU868_ED_MAC_BV_001_A
-- | -- | --
63 | Create: | Component: mod_core_TestSystem.mtc_TesterExecutable
1407 | Message: | Activation Over The Air (OTAA)
1563 | Message: | Over-The-Air Activation (OTAA) process starts. Waiting Join-request message...
1563 | StartTimer: | Start Timer t_TOwait, Duration: 90.00 s
1610 | Recv: | FRAME:= {push_data := {'01'O, '7A1D'O, '00'O, 'FFFEB827EB56B127'O, {{omit, omit, 714596924, 868.500000, 2, 1, 1, "LORA", "SF12BW125", "4/5", -90, 8.800000, 14, "QBlFmyyAEQACxMnDr18=", {'01000000'B, {macpayload_mac_message := {{'2C9B4519'O, '10000000'B, '1100'O, omit}, 2, 'C4'O}}, 'C9C3AF5F'O}}}}}
1610 | Message: | Uplink (Not a Join Request message) received
1625 | Send: | FRAME:= {push_ack := {'01'O, '7A1D'O, '01'O}}
3516 | Recv: | FRAME:= {pull_data := {'01'O, '8C86'O, '02'O, 'FFFEB827EB56B127'O}}
3532 | Send: | FRAME:= {pull_ack := {'01'O, '8C86'O, '04'O, 'FFFEB827EB56B127'O}}
6219 | Recv: | FRAME:= {push_data := {'01'O, 'A536'O, '00'O, 'FFFEB827EB56B127'O, {{omit, omit, 720600708, 868.100000, 0, 1, 1, "LORA", "SF12BW125", "4/5", -90, 10.200000, 14, "QBlFmyyAEgACSdVBS0M=", {'01000000'B, {macpayload_mac_message := {{'2C9B4519'O, '10000000'B, '1200'O, omit}, 2, '49'O}}, 'D5414B43'O}}}}}
6219 | Message: | Uplink (Not a Join Request message) received
6235 | Send: | FRAME:= {push_ack := {'01'O, 'A536'O, '01'O}}
9328 | Recv: | FRAME:= {push_data := {'01'O, 'D007'O, '00'O, 'FFFEB827EB56B127'O, {{omit, omit, 723688324, 868.500000, 2, 1, 1, "LORA", "SF12BW125", "4/5", -95, 10.200000, 23, "AAAREQAAEREAGUWbLKpVm2BprI5o13c=", {'00000000'B, {macpayload_join_request := {'0011110000111100'O, '609B55AA2C9B4519'O, '69AC'O}}, '8E68D777'O}}}}}
9328 | Message: | Uplink (Join Request message) received
9344 | Send: | FRAME:= {push_ack := {'01'O, 'D007'O, '01'O}}
9469 | Message: | Sending Join Accept message
9485 | Send: | FRAME:= {pull_resp := {'01'O, '0000'O, '03'O, {false, 729688324, 0, 869.525000, 0, 14, "LORA", "SF12BW125", "4/5", true, 17, "IIqLqcPgoCh/YtIMYBKRxbA=", false, false, 8, 0, false, false, 0}, {phypayload := {'00100000'B, {macpayload_join_accept := {'020000'O, '000000'O, '2C9B4519'O, '00'O, '01'O, omit}}, '84F2812D'O}}}}
13688 | Recv: | FRAME:= {pull_data := {'01'O, '71C0'O, '02'O, 'FFFEB827EB56B127'O}}
13719 | Send: | FRAME:= {pull_ack := {'01'O, '71C0'O, '04'O, 'FFFEB827EB56B127'O}}
21000 | Recv: | FRAME:= {push_data := {'01'O, '6D73'O, '00'O, 'FFFEB827EB56B127'O, {{omit, omit, 735368972, 868.100000, 0, 1, 1, "LORA", "SF12BW125", "4/5", -91, 10.200000, 14, "QBlFmyyAAQACh5MKElg=", {'01000000'B, {macpayload_mac_message := {{'2C9B4519'O, '10000000'B, '0100'O, omit}, 2, '87'O}}, '930A1258'O}}}}}
21000 | StopTimer: | Stop Timer t_TOwait
21000 | Message: | Uplink received from DUT
21016 | Send: | FRAME:= {push_ack := {'01'O, '6D73'O, '01'O}}
21032 | Message: | Activation (OTAA) process has finished successfully
21032 | Preliminary verdict: | Partial Verdict: PASS
21032 | Message: | Data stored in the DUT after Activation (OTAA):
21032 | Message: | DevAddr: 2C9B4519
21032 | Message: | AppEUI: 0011110000111100
21032 | Message: | NwkSKey: B8ADA5E02C092C55740B9D39A935520A
21032 | Message: | AppSKey: 38034B101F62BBC87C70ECCCFEDAC684
21032 | Message: | Test mode activation process starts...
21032 | StartTimer: | Start Timer t_TOwait, Duration: 90.00 s
23735 | Recv: | FRAME:= {pull_data := {'01'O, '4EF8'O, '02'O, 'FFFEB827EB56B127'O}}
23750 | Send: | FRAME:= {pull_ack := {'01'O, '4EF8'O, '04'O, 'FFFEB827EB56B127'O}}
27000 | Recv: | FRAME:= {push_data := {'01'O, 'C58C'O, '00'O, 'FFFEB827EB56B127'O, {{omit, omit, 741372764, 868.100000, 0, 1, 1, "LORA", "SF12BW125", "4/5", -95, 10.800000, 14, "QBlFmyyAAgACzAis4U0=", {'01000000'B, {macpayload_mac_message := {{'2C9B4519'O, '10000000'B, '0200'O, omit}, 2, 'CC'O}}, '08ACE14D'O}}}}}
27000 | Message: | Uplink received from DUT
27016 | Send: | FRAME:= {push_ack := {'01'O, 'C58C'O, '01'O}}
27032 | Message: | Global parameters updated
27172 | Message: | Sending Test mode activation message [FCnt: 1. FRMPayload (decrypted): 01010101]
27172 | Send: | FRAME:= {pull_resp := {'01'O, '0000'O, '03'O, {false, 743372764, 0, 869.525000, 0, 14, "LORA", "SF12BW125", "4/5", true, 17, "YBlFmywAAQDg38CbcdrbrC0=", false, false, 8, 0, false, false, 0}, {phypayload := {'01100000'B, {macpayload_mac_message := {{'2C9B4519'O, '00000000'B, '0100'O, omit}, 224, 'DFC09B71'O}}, 'DADBAC2D'O}}}}
[... 705 lines cut ...]
146328 | Message: | Unconfirmed mode tested successfully
146344 | Preliminary verdict: | Partial Verdict: FAIL
146344 | Message: | Test mode deactivation process starts...
146344 | StartTimer: | Start Timer t_TOwait, Duration: 90.00 s
151281 | Recv: | FRAME:= {push_data := {'01'O, '26ED'O, '00'O, 'FFFEB827EB56B127'O, {{omit, omit, 4208880996, 868.500000, 2, 1, 1, "LORA", "SF12BW125", "4/5", -88, 7.800000, 15, "QBlFmyyAHADgc9ukL8/t", {'01000000'B, {macpayload_mac_message := {{'2C9B4519'O, '10000000'B, '1C00'O, omit}, 224, '73DB'O}}, 'A42FCFED'O}}}}}
151281 | Message: | Uplink received from DUT
151297 | Send: | FRAME:= {push_ack := {'01'O, '26ED'O, '01'O}}
151312 | Message: | Global parameters updated
151437 | Message: | Sending Test mode deactivation message [FCnt: 107415. FRMPayload (decrypted): 00]
151453 | Send: | FRAME:= {pull_resp := {'01'O, '0000'O, '03'O, {false, 4209880996, 0, 868.500000, 0, 14, "LORA", "SF12BW125", "4/5", true, 14, "YBlFmywAl6Pge4k0k6g=", false, false, 8, 0, false, false, 0}, {phypayload := {'01100000'B, {macpayload_mac_message := {{'2C9B4519'O, '00000000'B, '97A3'O, omit}, 224, '7B'O}}, '893493A8'O}}}}
154625 | Recv: | FRAME:= {push_data := {'01'O, 'AA07'O, '00'O, 'FFFEB827EB56B127'O, {{omit, omit, 4212230436, 868.300000, 1, 1, 1, "LORA", "SF12BW125", "4/5", -89, 9.000000, 14, "QBlFmyyAHQACl4khZyE=", {'01000000'B, {macpayload_mac_message := {{'2C9B4519'O, '10000000'B, '1D00'O, omit}, 2, '97'O}}, '89216721'O}}}}}
154625 | Message: | Uplink received from DUT
154641 | Send: | FRAME:= {push_ack := {'01'O, 'AA07'O, '01'O}}
154656 | Message: | Global parameters updated
154687 | Message: | #FPort received: 2
154687 | Message: | #Decrypted Payload received: 0xFF
154687 | StopTimer: | Stop Timer t_TOwait
154687 | Message: | Test mode has been disabled successfully
154687 | Preliminary verdict: | Partial Verdict: FAIL
154703 | Test Case Ended: | mod_LORA_A_EU868.TC_A_EU868_ED_MAC_BV_006
154719 | Final verdict: | Final Verdict: FAIL

`

Thank You

pjb-loxius, 2020-12-22:
Bedankt voor uw email.
 
Ik ben afwezig tot maandag 4 januari. E-mail gericht aan dit e-mailadres zal niet tussentijds worden gelezen of worden doorgestuurd.

Dank voor uw begrip. 
Met vriendelijke groet,

ing. Peter Boudewijns
Loxius BV
p.boudewijns@loxius.nl

Loxius BV Office: +31-162-522052 / Fax: +31-162-519495
Lissenveld 47c, 4941 VL Raamsdonksveer, The Netherlands
http://www.loxius.nl

--------------------

Thank you for your e-mail. 

As we are moving to our new office I am not available to answer your e-mail on a short notice. On Tuesday, March 9th, I will be back in office and will answer your e-mail then. Your e-mail has not been forwarded.

Thank you for your understanding.

Kind regards,

ing. Peter Boudewijns
Loxius BV
p.boudewijns@loxius.nl

Loxius BV Office: +31-162-522052
Lissenveld 47c, 4941 VL Raamsdonksveer, The Netherlands
http://www.loxius.nl

mluis1, 2021-01-26:
Hope you don't mind, I have just edited your message in order to improve readability. (Please review the changes in case I have made a mistake)

I have just run the tests you mention as failing and on my side they all succeeded using  v4.4.1 release.
To perform the tests I used the following:
- NucleoL476+SX1261MBXBAS
- periodic-uplink-lpp example
- LCTT pre-certification tool v1.7.0 with technology package v3.1.0_D1

Please find attached a zip file containing the LCTT generated log files for the mentioned tests.
[github-issue-1017-lctt-logs.zip](https://github.com/Lora-net/LoRaMac-node/files/5874028/github-issue-1017-lctt-logs.zip)

Unfortunately, I cannot say why in your case these tests are failing. Also, it will be quite hard to help you figure out what could be the possible causes as I don't know which modifications you have done to the code and I don't know if the porting to the BlueNRG-2 platform is fully functional.

Although by analyzing the provided logs (which by the way don't look to have been generated by the LCTT tool) I may provide some clues on what you could look at in order to debug your system.

- BV001_A
  It looks like that the test fails because the random selection of frequencies did not gave the 868.5 MHz channel.
  I would recommend to run the test again. By running it again the random generated frequencies list may be different and all default EU868 channels could come out of it and the test to succeed.  
  You may also verify if your random generator is behaving as expected. 
- BV003
 I observed that in very rare occasions when running all tests in a single test session using the LCTT tool this test fails. But, running it a second time it would work. I never had the time to fully analyze why this happens but, I suspect that the test fails because of the execution of the previous test.  Maybe, the LCTT tool previous test execution sets the DUT in a configuration that doesn't work anymore as expected.
 I would also verify if the cryptographic related functions are all correctly implemented.
- BV006
 I never observed this issue. The only recommendation I can make is to verify if the cryptographic related functions are all correctly implemented.
- BV015_A
 I don't know what is the environment where you have run your tests.  
 Possible causes to have this test fail are multiple.  
  - Other LoRaWAN devices working close to the one being tested
  - Having the end-device too close to the gateway
  - Using an antenna on the gateway and on the end-device. It would be better to use a cable between the gateway and the end-device. Please do not forget to put the right attenuation on the cable.
  - etc...

Hope it helps you find what causes your issues.

AsteelDev, 2021-02-01:
Hello, 

Thank you for your feedback. We use LoRa Stack Release 4.4.5 but you tested with Release 4.4.1

Regards,

mluis1, 2021-02-02:
Sorry, I guess that I mixed up the versions.

I have run the tests for version 4.4.5. Please find attached a zip file containing the 4.4.5 generated logs.
I have also included the DUT serial port log.

[github-issue-1017-lctt-logs-4.4.5.zip](https://github.com/Lora-net/LoRaMac-node/files/5910308/github-issue-1017-lctt-logs-4.4.5.zip)

mluis1, 2021-03-03:
Can we close this issue?

mluis1, 2021-03-18:
No feedback, we close the issue.

closed 2021-03-18 by mluis1