Lora-net/LoRaMac-node #1019: First packet always contains empty frame

MedadRufus, 2020-12-26:
Under the following conditions, the first packet sent after boot is always an empty frame. Subsequent transmissions have the right payload.

- release V4.4.5
- setting `LORAWAN_ADR_STATE` to `LORAMAC_HANDLER_ADR_OFF`
- setting `LORAWAN_DEFAULT_DATARATE` to `DR_4`
- setting `ACTIVE_REGION` to `LORAMAC_REGION_EU868`
- payload size greater than 51 bytes

The cause appears to be due to be the 'CurrentPossiblePayloadSize`  being calculated for datarate DR0 during the first transmission, not the datarate DR4 that I set. My target packet size of 51+ bytes is too long for DR0 while its OK for DR4, hence the empty frame.

Further clues I have found are:

Before the first transmission, `LoRaMacQueryTxPossible(...)` is called and `adrNext.Datarate` is set to `MacCtx.NvmCtx->MacParams.ChannelsDatarate`. However, `MacCtx.NvmCtx->MacParams.ChannelsDatarate` is initialised to `0` and not `4` which is the default datarate I selected. Hence, further down, in line 3483, `CurrentPossiblePayloadSize` is incorrectly calculated to be only 51 bytes. If my packet is expected to be bigger than that, it returns `LORAMAC_STATUS_LENGTH_ERROR`.

https://github.com/Lora-net/LoRaMac-node/blob/118771fee14d5cfb3edbe4b0846a404df9002443/src/mac/LoRaMac.c#L3454-L3510

The correct datarate is only set later, only after the payload was set to an empty frame:
https://github.com/Lora-net/LoRaMac-node/blob/118771fee14d5cfb3edbe4b0846a404df9002443/src/mac/LoRaMac.c#L4786

I suggest in `LmHandler.c` in `LmHandlerInit(...)` to add in the following lines to initialise the datarate:

```
mibReq.Type = MIB_CHANNELS_DEFAULT_DATARATE;
mibReq.Param.ChannelsDefaultDatarate = LmHandlerParams->TxDatarate;
LoRaMacMibSetRequestConfirm( &mibReq );

mibReq.Type = MIB_CHANNELS_DATARATE;
mibReq.Param.ChannelsDatarate = LmHandlerParams->TxDatarate;
LoRaMacMibSetRequestConfirm( &mibReq );
```

It works for me and the first packet now is not empty.

mluis1, 2021-03-03:
Thanks for the finding. We will investigate how to best solve the issue based on your proposition.

mluis1, 2021-03-17:
@MedadRufus Could you please review?

MedadRufus, 2021-03-17:
Thank you @mluis1, I confirm the issue solved. Shall I close this issue?

mluis1, 2021-03-18:
Thanks for the feedback. Yes, please close it.

closed 2021-03-18 by MedadRufus