Lora-net/LoRaMac-node #1024: About LoRa channel activity detection in sx126x

yufu186, 2021-01-19:
Hi,
In sx126x.c, SX126xSetCadParams is proposed to set the parameters. However, the meaning of cadDetPeak and cadDetMin is not well explained. The datasheet of semtech only gives a vague explanation about the "sensitivity". So, What's the real function and meaning of those two parameters? Otherwise, Where does LoRa perform a correlation calculation in CAD? Can we control the calculation process? 
 

Thanks.

mluis1, 2021-01-19:
Looking at SX126x page on Semtech web site you will find an application note describing how to use the CAD.
[LoRa Channel Activity Detection (CAD) with SX126x](https://semtech.my.salesforce.com/sfc/p/#E0000000JelG/a/2R000000Q1ES/SPexo9njbhEQLcJVUg1i0Su8p3tpAtwX1jhMBGlXsQI)

In the future it would be nice if you could post this kind of questions on the project [Discussions](https://github.com/Lora-net/LoRaMac-node/discussions) tab. It is a better place to engage discussions and then we can agree if it is an issue or not.
If you don't mind I move this issue to the [Discussions](https://github.com/Lora-net/LoRaMac-node/discussions) tab.

yufu186, 2021-01-19:
OK, thanks for your reply!

yufu186, 2021-01-19:
> Looking at SX126x page on Semtech web site you will find an application note describing how to use the CAD.
> [LoRa Channel Activity Detection (CAD) with [...]

OK, thanks for your reply!

closed 2021-01-19 by mluis1