Lora-net/LoRaMac-node #1038: FUOTA Fails for large files

vbachler, 2021-02-05:
When sending larger files (above 2040 fragments) over FUOTA, the process will fail with an infinite for-loop.

In file FragDecoder.c line: 592
https://github.com/Lora-net/LoRaMac-node/blob/509a9e962d583b4e25342eaa1127349e870c071b/src/apps/LoRaMac/common/LmHandler/packages/FragDecoder.c#L592

The number of frames is passed through the `uint32_t` variable `m`. If the number of frames is above 2040, the expression 
`((m>>3) + 1)` is  above 256. Since `for loop` counting variable `i` is of `uint8_t` type, overflow occurs, and `for` becomes an infinite loop.

**Solution:** Simply change the type of variable `i` into `uint16_t`

mluis1, 2021-03-03:
Thanks for the finding. It will be corrected for the next release.

mluis1, 2021-03-10:
I have pushed the fix.