Lora-net/LoRaMac-node #1047: CLass B pingSlot frequency for AS923-2

denchii, 2021-02-17:
Greetings! We are working with class B (periodic-uplink-lpp sub-project) and found one value that seems not consistent with RP002-1.0.1 Regional Parameters:
The document defines AS923 pingSlot frequency = 923.4 MHz + AS923_FREQ_OFFSET_HZ, which is equal 921.6 MHz.
[image]
But the [RegionAS923.c](https://github.com/Lora-net/LoRaMac-node/blob/3d940d323f80c093d54c118ca978cc7730ee5ec5/src/mac/region/RegionAS923.c#L326-L330) doesn't take into account the frequency offset. Could you, please, tell me whether it was done intentionally and has some different logic behind it or it's actually wrong value? We could fix it in place but need some clarification.

mluis1, 2021-03-03:
Thanks for the finding. It will be corrected for the next release.

closed 2021-03-10 by mluis1

mluis1, 2021-03-10:
@denchii I have pushed the fix. Could you please check?