Lora-net/LoRaMac-node #1059: sx126x radio.c

RomanJasmann, 2021-03-12:
1. The critical section in RadioIrqProcess is not applied properly:

[image]

Fix:

```
void RadioIrqProcess(void) {
    uint8_t fired;

    CRITICAL_SECTION_BEGIN( );
    fired = IrqFired;
    IrqFired = 0;
    CRITICAL_SECTION_END( );

    if (fired == true) {

```

2. There is a missing TimerStop for the RxTimeoutTimer:

[image]

Fix: 

```
        if( ( irqRegs & IRQ_RX_DONE ) == IRQ_RX_DONE )
        {
            TimerStop( &RxTimeoutTimer );

            if( ( irqRegs & IRQ_CRC_ERROR ) == IRQ_CRC_ERROR )
            {
                if( RxContinuous == false )
                {
                    //!< Update operating mode state to a value lower than \ref MODE_STDBY_XOSC
                    SX126xSetOperatingMode( MODE_STDBY_RC );
                }
                if( ( RadioEvents != NULL ) && ( RadioEvents->RxError ) )
                {
                    RadioEvents->RxError( );
                }
            }
            else
            {
                uint8_t size;

                if( RxContinuous == false )
                {
                    //!< Update operating mode state to a value lower than \ref MODE_STDBY_XOSC
                    SX126xSetOperatingMode( MODE_STDBY_RC );

                    // WORKAROUND - Implicit Header Mode Timeout Behavior, see DS_SX1261-2_V1.2 datasheet chapter 15.3
                    // RegRtcControl = @address 0x0902
                    SX126xWriteRegister( 0x0902, 0x00 );
                    // RegEventMask = @address 0x0944
                    SX126xWriteRegister( 0x0944, SX126xReadRegister( 0x0944 ) | ( 1 << 1 ) );
                    // WORKAROUND END
                }
                SX126xGetPayload( RadioRxPayload, &size , 255 );
                SX126xGetPacketStatus( &RadioPktStatus );
                if( ( RadioEvents != NULL ) && ( RadioEvents->RxDone != NULL ) )
                {
                    RadioEvents->RxDone( RadioRxPayload, size, RadioPktStatus.Params.LoRa.RssiPkt, RadioPktStatus.Params.LoRa.SnrPkt );
                }
            }
        }
```

3. In case of a header error, the RxError callback should be called:

[image]

Fix:

```
        if( ( irqRegs & IRQ_HEADER_ERROR ) == IRQ_HEADER_ERROR )
        {
            TimerStop( &RxTimeoutTimer );
            if( RxContinuous == false )
            {
                //!< Update operating mode state to a value lower than \ref MODE_STDBY_XOSC
                SX126xSetOperatingMode( MODE_STDBY_RC );
            }
            if( ( RadioEvents != NULL ) && ( RadioEvents->RxTimeout != NULL ) )
            {
                RadioEvents->RxError( );
            }
        }
```

mluis1, 2021-03-12:
Thanks for the findings.

We agree with first two findings.

Concerning third finding we think that the implementation is correct as it is.

The `IRQ_HEADER_ERROR` can be seen as an RxTimemout.
This interrupt may happen at the beginning of the reception and as such it is similar to the RxTimeout.  
The RxTimeout corresponds to the amount of time required to receive the LoRa header.  
When the `IRQ_HEADER_ERROR` is generated the radio cannot anyway continue the reception process.

The RxError callback is more to notify the upper layer that there was a CRC error on the received frame.

mluis1, 2021-03-15:
@RomanJasmann Could you please review the proposition?

RomanJasmann, 2021-03-15:
@mluis1 you have fixed it.

closed 2021-03-15 by mluis1