Lora-net/LoRaMac-node #1060: SX126x: IrqFired potentially gets stuck in state false

rdaforno, 2021-03-15:
If a radio interrupt occurs between `SX126xGetIrqStatus()` and `SX126xClearIrqStatus()` in [`RadioIrqProcess()`](https://github.com/Lora-net/LoRaMac-node/blob/master/src/radio/sx126x/radio.c#L1223), the RF_DIO1 pin won't drop and thus no MCU interrupt will be triggered. As a consequence, `IrqFired` will be stuck in the state `false` permanently.

Possible solution: either poll the radio IRQ flags in a while loop until the DIO1 line drops or do not set `IrqFired` to `false` if the DIO1 is still high after sending `SX126xClearIrqStatus()`.

mluis1, 2021-03-16:
The chances for this situation to happen are quite low as radio interrupts don't really happen at the same time.
Nevertheless in order to improve robustness we should find a better way to handle this situation.

rdaforno, 2021-03-16:
Depending on the selected SPI frequency, the window for this to happen is actually several tens of microseconds (in our case it was ~20us). Depending on when `RadioIrqProcess()` is called, this issue is actually not that unlikely to occur unless I've missed something.

mluis1, 2021-03-16:
We will try to figure out the best way to improve the situation.

We think for the moment that there are 4 possibilities to solve this:
1. Poll IRQ flags in a while loop. (your proposition)
    The use of a while loop looks to be dangerous as the MCU could potentially get stuck.
2. Check DIO1 after `SX126xClearIrqStatus()`  (your proposition)
    Looks to be a good candidate however the same reported problem could arise between the call to `SX126xClearIrqStatus()` and DIO1 check. However the chances to see the problem will be greatly reduced.
3. Move `SX126xGetIrqStatus()` and `SX126xClearIrqStatus()` inside the critical section
   With this solution we ensure that the MCU doesn't generate an IRQ. Although, if the SPI driver relies on DMA transfers it will also prevent the DMA IRQs to be handled.
4. Move `SX126xGetIrqStatus()` and `SX126xClearIrqStatus()` to the IRQ handling callback
   This solution is similar to point 3. The only difference would be if the MCU supports nested interrupts which means that the DMA IRQ could still be handled.
5. Other solutions that we don't see for the moment.

rdaforno, 2021-03-16:
I think (2) will completely eliminate the problem if checking DIO1 and setting `IrqFired` is executed atomically. Consider the following:
- No interrupt comes between `SX126xClearIrqStatus()` and DIO1 check, but DIO1 is still high -> `IrqFired` will be set to `true` (correct)
- No interrupt comes between `SX126xClearIrqStatus()`and DIO1 check, and DIO1 is low -> `IrqFired` will be set to `false` (correct)
- An interrupt arrives between `SX126xClearIrqStatus()` and DIO1 check, then DIO1 will be read as high -> `IrqFired` will be set to `true` (correct)
- An interrupt arrives just after checking the DIO1 pin -> `IrqFired` will be set to false, but the interrupt will still be triggered and handled by the MCU later (correct)

(3) won't work. The problem is not that an interrupt preempts the execution of `SX126xGetIrqStatus()` or `SX126xClearIrqStatus()`, it is the fact that the DIO1 line might not drop low and since interrupts on the MCU are edge triggered, we miss it.

(4) wont work either

mluis1, 2021-03-16:
Personally, I also prefer solution 2.
Concerning 3. and 4. I'm not in full agreement with you. Anyway solution 2 is the best one.

would you be OK with below patch? It implements proposition 2.

patch file:
<details>
  <summary>Click to expand!</summary>
<p>

```patch
diff --git a/src/boards/NucleoL073/sx1261mbxbas-board.c b/src/boards/NucleoL073/sx1261mbxbas-board.c
index e53a88c5..ab66e57d 100644
--- a/src/boards/NucleoL073/sx1261mbxbas-board.c
+++ b/src/boards/NucleoL073/sx1261mbxbas-board.c
@@ -330,6 +330,11 @@ bool SX126xCheckRfFrequency( uint32_t frequency )
     return true;
 }
 
+uint32_t SX126xGetDio1PinState( void )
+{
+    return GpioRead( &SX126x.DIO1 );
+}
+
 #if defined( USE_RADIO_DEBUG )
 static void SX126xDbgPinTxWrite( uint8_t state )
 {
diff --git a/src/boards/NucleoL073/sx1262mbxcas-board.c b/src/boards/NucleoL073/sx1262mbxcas-board.c
index 8482f657..511d4ffb 100644
--- a/src/boards/NucleoL073/sx1262mbxcas-board.c
+++ b/src/boards/NucleoL073/sx1262mbxcas-board.c
@@ -330,6 +330,11 @@ bool SX126xCheckRfFrequency( uint32_t frequency )
     return true;
 }
 
+uint32_t SX126xGetDio1PinState( void )
+{
+    return GpioRead( &SX126x.DIO1 );
+}
+
 #if defined( USE_RADIO_DEBUG )
 static void SX126xDbgPinTxWrite( uint8_t state )
 {
diff --git a/src/boards/NucleoL073/sx1262mbxdas-board.c b/src/boards/NucleoL073/sx1262mbxdas-board.c
index 2181a6a8..fc450545 100644
--- a/src/boards/NucleoL073/sx1262mbxdas-board.c
+++ b/src/boards/NucleoL073/sx1262mbxdas-board.c
@@ -337,6 +337,11 @@ bool SX126xCheckRfFrequency( uint32_t frequency )
     return true;
 }
 
+uint32_t SX126xGetDio1PinState( void )
+{
+    return GpioRead( &SX126x.DIO1 );
+}
+
 #if defined( USE_RADIO_DEBUG )
 static void SX126xDbgPinTxWrite( uint8_t state )
 {
diff --git a/src/boards/NucleoL152/sx1261mbxbas-board.c b/src/boards/NucleoL152/sx1261mbxbas-board.c
index e53a88c5..ab66e57d 100644
--- a/src/boards/NucleoL152/sx1261mbxbas-board.c
+++ b/src/boards/NucleoL152/sx1261mbxbas-board.c
@@ -330,6 +330,11 @@ bool SX126xCheckRfFrequency( uint32_t frequency )
     return true;
 }
 
+uint32_t SX126xGetDio1PinState( void )
+{
+    return GpioRead( &SX126x.DIO1 );
+}
+
 #if defined( USE_RADIO_DEBUG )
 static void SX126xDbgPinTxWrite( uint8_t state )
 {
diff --git a/src/boards/NucleoL152/sx1262mbxcas-board.c b/src/boards/NucleoL152/sx1262mbxcas-board.c
index 8482f657..511d4ffb 100644
--- a/src/boards/NucleoL152/sx1262mbxcas-board.c
+++ b/src/boards/NucleoL152/sx1262mbxcas-board.c
@@ -330,6 +330,11 @@ bool SX126xCheckRfFrequency( uint32_t frequency )
     return true;
 }
 
+uint32_t SX126xGetDio1PinState( void )
+{
+    return GpioRead( &SX126x.DIO1 );
+}
+
 #if defined( USE_RADIO_DEBUG )
 static void SX126xDbgPinTxWrite( uint8_t state )
 {
diff --git a/src/boards/NucleoL152/sx1262mbxdas-board.c b/src/boards/NucleoL152/sx1262mbxdas-board.c
index 2181a6a8..fc450545 100644
--- a/src/boards/NucleoL152/sx1262mbxdas-board.c
+++ b/src/boards/NucleoL152/sx1262mbxdas-board.c
@@ -337,6 +337,11 @@ bool SX126xCheckRfFrequency( uint32_t frequency )
     return true;
 }
 
+uint32_t SX126xGetDio1PinState( void )
+{
+    return GpioRead( &SX126x.DIO1 );
+}
+
 #if defined( USE_RADIO_DEBUG )
 static void SX126xDbgPinTxWrite( uint8_t state )
 {
diff --git a/src/boards/NucleoL476/sx1261mbxbas-board.c b/src/boards/NucleoL476/sx1261mbxbas-board.c
index e53a88c5..ab66e57d 100644
--- a/src/boards/NucleoL476/sx1261mbxbas-board.c
+++ b/src/boards/NucleoL476/sx1261mbxbas-board.c
@@ -330,6 +330,11 @@ bool SX126xCheckRfFrequency( uint32_t frequency )
     return true;
 }
 
+uint32_t SX126xGetDio1PinState( void )
+{
+    return GpioRead( &SX126x.DIO1 );
+}
+
 #if defined( USE_RADIO_DEBUG )
 static void SX126xDbgPinTxWrite( uint8_t state )
 {
diff --git a/src/boards/NucleoL476/sx1262mbxcas-board.c b/src/boards/NucleoL476/sx1262mbxcas-board.c
index 8482f657..511d4ffb 100644
--- a/src/boards/NucleoL476/sx1262mbxcas-board.c
+++ b/src/boards/NucleoL476/sx1262mbxcas-board.c
@@ -330,6 +330,11 @@ bool SX126xCheckRfFrequency( uint32_t frequency )
     return true;
 }
 
+uint32_t SX126xGetDio1PinState( void )
+{
+    return GpioRead( &SX126x.DIO1 );
+}
+
 #if defined( USE_RADIO_DEBUG )
 static void SX126xDbgPinTxWrite( uint8_t state )
 {
diff --git a/src/boards/NucleoL476/sx1262mbxdas-board.c b/src/boards/NucleoL476/sx1262mbxdas-board.c
index 2181a6a8..fc450545 100644
--- a/src/boards/NucleoL476/sx1262mbxdas-board.c
+++ b/src/boards/NucleoL476/sx1262mbxdas-board.c
@@ -337,6 +337,11 @@ bool SX126xCheckRfFrequency( uint32_t frequency )
     return true;
 }
 
+uint32_t SX126xGetDio1PinState( void )
+{
+    return GpioRead( &SX126x.DIO1 );
+}
+
 #if defined( USE_RADIO_DEBUG )
 static void SX126xDbgPinTxWrite( uint8_t state )
 {
diff --git a/src/boards/sx126x-board.h b/src/boards/sx126x-board.h
index c29291a8..d5e04f6c 100644
--- a/src/boards/sx126x-board.h
+++ b/src/boards/sx126x-board.h
@@ -159,6 +159,13 @@ bool SX126xCheckRfFrequency( uint32_t frequency );
  */
 uint32_t SX126xGetBoardTcxoWakeupTime( void );
 
+/*!
+ * \brief Gets current state of DIO1 pin state.
+ *
+ * \retval state DIO1 pin current state.
+ */
+uint32_t SX126xGetDio1PinState( void );
+
 /*!
  * \brief Gets the current Radio OperationMode variable
  *
diff --git a/src/radio/sx126x/radio.c b/src/radio/sx126x/radio.c
index 8939e432..3a537234 100644
--- a/src/radio/sx126x/radio.c
+++ b/src/radio/sx126x/radio.c
@@ -1260,6 +1260,14 @@ void RadioIrqProcess( void )
         uint16_t irqRegs = SX126xGetIrqStatus( );
         SX126xClearIrqStatus( irqRegs );
 
+        // Check if DIO1 pin is High. If it is the case revert IrqFired to true
+        CRITICAL_SECTION_BEGIN( );
+        if( SX126xGetDio1PinState( ) == 1 )
+        {
+            IrqFired = true;
+        }
+        CRITICAL_SECTION_END( );
+
         if( ( irqRegs & IRQ_TX_DONE ) == IRQ_TX_DONE )
         {
             TimerStop( &TxTimeoutTimer );
```

</p>
</details>

rdaforno, 2021-03-16:
The patch looks fine.

However, I think it is important to understand the issue before implementing a fix. If we don't agree that the proposed solution (3) and (4) won't work, then we have a different understanding of the problem.
Let me try to illustrate the issue with the following diagram. The first part shows normal operation where the DIO1 pin drops low after sending the clear command, whereas the 2nd part shows the erratic behaviour where the line stays high because another radio event has arrived in the meantime:

[image]

A critical section won't prevent such a condition.

mluis1, 2021-03-16:
Ok, I see your point concerning proposal 3.
As the MCU IRQ is only generated on a rising edge we may miss it if happens in the critical section.

However, for point 4 the clearing is handled on the `DIO1 ISR` callback thus avoiding potential additional delays. 
Also, the chances that another radio IRQ happens while handling one already being processed is small.
This solution has other drawbacks (SPI DMA usage) anyway.

In the mean time I have proposed the provided patch to our internal systems and waiting on my colleagues feedback.

mluis1, 2021-03-29:
@rdaforno I have just pushed the fix for this issue.
Could you please review it and close the issue?

closed 2021-03-31 by rdaforno

RomanJasmann, 2021-09-14:
> 
> 
> @rdaforno I have just pushed the fix for this issue.
> Could you please review it and close the issue?

What is the purpose of the critical section in the fix? 

```
        // Check if DIO1 pin is High. If it is the case revert IrqFired to true
        CRITICAL_SECTION_BEGIN( );
        if( SX126xGetDio1PinState( ) == 1 )
        {
            IrqFired = true;
        }
        CRITICAL_SECTION_END( );
```

rdaforno, 2021-09-14:
@RomanJasmann This is more of a workaround, not a real solution. Potentially loosing events may be acceptable in some cases, but it is certainly not the expected / wanted behaviour in general.