Lora-net/LoRaMac-node #1065: Multicast Channel Setup RxParams issue in return status

sjain315, 2021-03-18:
LoRaMacMcChannelSetupRxParams return wrong status if frequency or Datarate verification get failed.

If wrong frequency is used to set in LoRaMacMcChannelSetupRxParams then it should return LORAMAC_STATUS_FREQUENCY_INVALID instead of LORAMAC_STATUS_OK.

 
[image]

mluis1, 2021-03-18:
There is no error on this function implementation.

This function is parsing the command received from the  server and prepares the answer parameters to be sent back to the server.

The data rate and channel frequency verification has nothing to do with function execution returned status.

sjain315, 2021-03-19:
> The data rate and channel frequency verification has nothing to do with function execution returned status.

Return status of that function is used to decide whether need to start multicast session or not.
you can check that callback in LmhpRemoteMcastSetup.c file.
In LmhpRemoteMcastSetupOnMcpsIndication function there is a case REMOTE_MCAST_SETUP_MC_GROUP_CLASS_C_SESSION_REQ.
In that case if we receive LORAMAC_STATUS_OK as a return for invalid data rate and channel frequency than also it will start multicast classC session with wrong parameters.

_Correct me please If I misunderstood something but I found this case so reported here._

[image]

mluis1, 2021-07-07:
I see your point.

Up until we don't fix this I propose to change `LoRaMacMcChannelSetupRxParams` function implementation to something like:

```c
LoRaMacStatus_t LoRaMacMcChannelSetupRxParams( AddressIdentifier_t groupID, McRxParams_t *rxParams, uint8_t *status )
{

...

    if( *status == ( groupID & 0x03 ) )
    {
        // Apply parameters
        Nvm.MacGroup2.MulticastChannelList[groupID].ChannelParams.RxParams = *rxParams;
    }
    else
    {
        return LORAMAC_STATUS_PARAMETER_INVALID;
    }

    return LORAMAC_STATUS_OK;
}
```

closed 2021-12-09 by mluis1