Lora-net/LoRaMac-node #1066: Context management restores soft-se root keys?

brocaar, 2021-03-19:
It looks like when `CONTEXT_MANAGEMENT_ENABLED` is enabled, together with the soft-se option, that changing the keys in `se-identity.h` has no effect. I noticed this after updating the `LORAWAN_JOIN_EUI` and `NWK_KEY`. After (re)flashing, the device kept using the old `LORAWAN_JOIN_EUI` (and I assume as well the `NWK_KEY`, as the device failed to join).

I tried removing the `build` folder, `make clean` etc... but none of my changes had any effect. Then I noticed the `###### =========== CTXS RESTORED =========== ######` message. After disabling `CONTEXT_MANAGEMENT_ENABLED`, rebuild and reflash, the device succesfully joined using the new `LORAWAN_JOIN_EUI` and `NWK_KEY`.

mluis1, 2021-03-19:
A few questions:
- Which version are you using?
- Which platform?
- Have you tried to make a mass-erase of the MCU Flash memory?
   The NVM data memory doesn't not get erased by updating the firmware.

If you are using a version newer or equal to v4.4.7 (LoRaWAN 1.0.3), v.4.5.1 (LoRaWAN 1.0.4) or master (1.0.4)/develop(1.0.4/1.1.1) branch you have the possibility to erase the NVM data through the serial interface by pressing ESC then shift N keys.

mluis1, 2021-03-19:
If you don't mind I move this issue to the [Discussions](https://github.com/Lora-net/LoRaMac-node/discussions) tab.

closed 2021-03-19 by mluis1