Lora-net/LoRaMac-node #1086: sysIrqHandlers.c for cortex-m0

dudmuck, 2021-04-21:
for users of cortex-m0 devices (ie stm32L0), might be useful to have correct hardfault handler, for example for gcc-arm:

```
void HardFault_Handler(void)
{
    /*
    __asm volatile( "TST LR, #4" );
    __asm volatile( "ITE EQ" );
    __asm volatile( "MRSEQ R0, MSP" );
    __asm volatile( "MRSNE R0, PSP" );
    __asm volatile( "B HardFault_Handler_C" );
    */

    // This version is for Cortex M0
    __asm volatile("movs R0, #4");
    __asm volatile("mov R1, LR");
    __asm volatile("tst R0, R1"); // Check EXC_RETURN in Link register bit 2.
    __asm volatile("bne Uses_PSP");
    __asm volatile("mrs R0, MSP");// Stacking was using MSP.
    __asm volatile("b Pass_StackPtr");
    __asm volatile("Uses_PSP:");
    __asm volatile("mrs R0, PSP");  // Stacking was using PSP
    __asm volatile("Pass_StackPtr:");
    __asm volatile("ldr R2,=HardFault_Handler_C");
    __asm volatile("bx  R2");
    __asm volatile("");
}
```

see at https://www.segger.com/downloads/application-notes/AN00016 
its because Cortex-M0 doesnt have "TST LR, #4" and "ITE EQ".

Could be useful for debugging hard fault on cortex-m0.

mluis1, 2022-10-21:
@dudmuck Could you please review the attached patch?
Please let us know if you are OK with the proposition.

[m0_hard_fault_handler_fix.patch.txt](https://github.com/Lora-net/LoRaMac-node/files/9839249/m0_hard_fault_handler_fix.patch.txt)

dudmuck, 2022-10-21:
i compiled that, its good.  But only builds when `HARD_FAULT_HANDLER_ENABLED` defined.  Perhaps the readme needs a note on how to enable it.

closed 2022-10-25 by mluis1