Lora-net/LoRaMac-node #1103: V4.4.1 class C downlink doesn't work at all.

Newsland, 2021-06-01:
Hi,guys.I'm sorry to trouble you that I move LoRaMac node to my board(stm32l431cbt6 with 32.768khz LSE,8Mhz HSE).and the class A mode works very well.But when I switch to class C mode.my board can't receive downlink packet.Then I do these jobs to check anything wrong:
1.ensure gateway and node both switch to class c mode,same frequency,datarate,coderate,bandwith,spreadfactor.
2.ensure gateway works fine(can receive and send lora packets)
3.ensure node works and open continue rx2 window.
4.I enter debug and set broke points.and find it didn't enter RX_DONE section after I send a downlink packet.Then I found others has the same problem and you update this issue,so I download the new files(sx126x.h sx126x.c  radio.c) and replace,but it can't work.Could you please give me some advices.
Thank you a lot.

Newsland, 2021-06-02:
I use open source lora server,other device with lora module can work,and we are trying to replace lora module by using chips.Actually the problem is rx2 windows period can't receive data whatever in class A or class C.And our lora server can distinguish class a and class c,because we configure basic information on it.And we only use ABP.
[image]

Newsland, 2021-06-02:
[image]

Newsland, 2021-06-02:
I found issue#498 has the same problem,then update this files.But not works

altishchenko, 2021-06-02:
Hello and what this ACK-False thing is? And what is marked in blue? I am not familiar with this server, so I can't tell where is uplink and where is downlink and for what device this downlink is destined if blue is downlink. For what I can see is that the server receives information from your device and sends nothing back (?) You restart your device often and use ABP with that, what is the setting of the FCnt check on the server side? Some servers will discard "duplicate" FCnts from the devices if not instructed otherwise. Devices too may discard unknown FCnts from the server.. For example, your device doesn't store session data (FCnts) in the NVRAM, so upon its restart it will expect and FCnt of 1 or 0, but your server may remember trying to send some data down to your device and will choose its own next value, not 1 or 0. Your device will naturally ignore this packet as the FCnt count is incorrect / unexpected.

Pirakutu, 2021-06-03:
The stack had issues running class C on that version (4.4.2 also), i also had the same problems when running it on ABP. I would recommend to update the stack to the newest version, or at least to V4.4.4 (the stack itself works well on class C there, we are running it there and it works very stable). There was only a problem with the radio drivers on the sx126x side on that version of the stack, so we updated the radio files to v4.4.7, which then solved the problem. Hope this helps.

closed 2021-06-03 by mluis1