Lora-net/LoRaMac-node #1109: Listen Before Talk with 128 us scan time does not allow tx of 400ms duration

reissjason, 2021-06-17:
ARIB STD-T108 Version 1. 3 Lists 400ms tx duration from start of transmission is allowed.
The packet forwarder is counting time between channel was free and start of transmission in delta time.
Once delta time passes 400ms the tx is not allowed.

tx_mode                    = TIMESTAMPED
lbt_is_channel_free:273: LBT: select channel 1 (923400000 Hz)
lbt_is_channel_free:333: sx1301_time                = 3682304
lbt_is_channel_free:334: tx_freq                    = 923400000
------------------------------------------------
lbt_is_channel_free:336: packet_duration            = 370000
lbt_is_channel_free:337: tx_start_time              = 2285568
lbt_is_channel_free:338: lbt_time1                  = 2232320
lbt_is_channel_free:339: lbt_time2                  = 0
lbt_is_channel_free:340: lbt_time                   = 2232320
lbt_is_channel_free:341: delta_time                 = 421888
------------------------------------------------
ERROR: TX request rejected (LBT)

mluis1, 2021-06-18:
@reissjason If I am not wrong I guess that you have posted this issue on the wrong project.
It looks like to be an issue related to a gateway isn't it?

mluis1, 2021-06-22:
@reissjason Could you please close this issue?

reissjason, 2021-06-22:
Yes wrong project, sorry.

closed 2021-06-22 by reissjason