Lora-net/LoRaMac-node #1118: NRF52840DK with sx1262mb2xAS shield  - MCPS CONFIRM status: Rx 2 timeout

cawahe, 2021-06-24:
Are there any solutions here for getting an error : MCPS CONFIRM status: Rx 2 timeout for confirmed messages. 
It is missing downlinks too when sending and receiving at the same time. 

Unconfirmed messages seems to be ok. but they never receive downlinks.   
Class A, downlinks on port 1

rfenderStrata, 2021-07-05:
Hello cawahe we are currently trying to adapt LoRaWAN for NRF52840 too, did you get your error from your own implementation of Lora-net/LoRaMac-node or are you using  this repo [nrf52-lora](https://github.com/gluedig/nrf52-lora) 
?

cawahe, 2021-07-06:
> Hello cawahe we are currently trying to adapt LoRaWAN for NRF52840 too, did you get your error from your own implementation of Lora-net/LoRaMac-node or are you using this repo [nrf52-lora](https://github.com/gluedig/nrf52-lora)
> ?

From this example. [https://github.com/FreeRTOS/Lab-Project-FreeRTOS-LoRaWAN](https://github.com/FreeRTOS/Lab-Project-FreeRTOS-LoRaWAN),  but seems like sending sends on all freq even after join. How can I alter that?

closed 2021-07-06 by mluis1