Lora-net/LoRaMac-node #1153: Porting to STM32CubeIDE

kvkc97, 2021-08-07:
I would like to port LoRaWAN Stack for a custom board built using STM32WB55RGV6 controller and LR1110 driver. I am using STM32CubeIDE for the same. Can I port and use the stack in STM32CubeIDE?

Pirakutu, 2021-08-13:
yes you can, i have a version of the stack running on a stm32wb55, with the LR1110, on CubeIDE. It works well.

Rgrndaal, 2021-08-13:
Bedankt voor uw email.
Op dit moment ben ik niet aanwezig op kantoor en heb niet altijd toegang tot mijn email.
Tijdens mijn afwezigheid kunt u het beste contact opnemen met Nick van Gils op 0162-511517 of e-mail ***@***.***

Thanks for your e-mail. 
I'm currently out of the office with limited access to my e-mail.
In my absence, please feel free to contact Nick van Gils on +31-(0)162-511517 or e-mail ***@***.***

Met vriendelijke groet,
Kind regards,

Remco Groenendaal
CLICKEY solutions
***@***.***

CLICKEY solutions Office: +31-162-511517
Lissenveld 47c, 4941 VL Raamsdonksveer, The Netherlands
www.clickey.eu

kvkc97, 2021-08-15:
@Pirakutu I have ported the stack to CubeIDE to work with STM32WB55RG controller. The RF communication was checked with LR1110 IC and the controller and packet was received at the gateway. However, after including the MAC layer in the code, the code gets stuck at SPI transmission  command. Could you please share your code so that I can use it as a reference?

Pirakutu, 2021-08-16:
What do you mean after including the mac layer? 

Does it get stuck at the busy pin or at an SPI command? As these are 2 different things.

kvkc97, 2021-08-16:
I have tested the LR1110 IC with STM32WB55RG controller using STM32CubeIDE and packet was received at the gateway. Afterwords I added Mac layer ( Mac, peripherals, LoRaWAN folder from apps) and made necessary changes in boards folder for STM32WB55RG controller and called the Lmhandler functions in main.cpp as given in this example. On running the code, it gets stuck at SPI transmission (it works well when communicating directly using IC). Busy pin is working fine and is set low as required.

kvkc97, 2021-09-07:
I have updated the code for SPI and the code is not getting stuck. However some other errors are displayed.
I was testing for LoRa transmission with LR1110 IC in OTA mode using LoRaMAC code. The Join request is successfully sent to the server and Join accept is generated from the server and is passed to the gateway. The gateway also forwards the packet. However the packet is not received at the end device (custom board developed using STM32 controller). I have attached the Join request and join accept messages.
The IC is going to rx mode after transmitting but packet is not received. At time the IC goes to busy after TX mode and gets stuck. At times it goes to RX mode. A few times when in RX mode, interrupt for preamble detection is generated but no further packet is received and most of the times preamble detection also is not done. What would be the issue? Is it due to the PLL error generated.
Join request at server
[image]
Join accept at server
[image]
Join request and Join Request at gateway
[image]

Pirakutu, 2021-09-07:
which LR1110 driver files did you use? i had problems using the newest ones. Had to go back to version 1.0.0.

kvkc97, 2021-09-07:
I used the latest version given in the following link
https://github.com/Lora-net/lr1110_driver
Did you go back to version 1.0.0 of the same driver?

Pirakutu, 2021-09-07:
Yeah, i went back to the version 1.0.0 of that same driver (i saw that was the version included in this project and hoped it worked that way).

kvkc97, 2021-09-08:
Hi, I reverted to version 1.0.0 and tried the code. The errors are still present

closed 2021-09-09 by mluis1