Lora-net/LoRaMac-node #1220: wrong condition in ConvertRejoinCycleTime?

sborshch, 2021-12-11:
function returns success on overflow only.
```
    // Our timer implementation do not allow longer times than 4294967295 ms
    if( rejoinCycleTime > 4294967 )
    {
        *timeInMiliSec = rejoinCycleTime * 1000;
        return true;
    }
    else
    {
        return false;
    }

```
seems like correct condition should be  `if( rejoinCycleTime <= 4294967 )`

pjb-loxius, 2021-12-11:
Bedankt voor uw email.
 
Ik ben afwezig tot maandag 10 januari. E-mail gericht aan dit e-mailadres zal niet tussentijds worden gelezen of worden doorgestuurd.

Dank voor uw begrip. 

Met vriendelijke groet,

ing. Peter Boudewijns
CLICKEY solutions
***@***.***

CLICKEY solutions: +31-162-522052 / Fax: +31-162-519495
Lissenveld 47c, 4941 VL Raamsdonksveer, The Netherlands
http://www.clickey.eu

--------------------

Thank you for your e-mail. 

I am out of office until Monday January 10th. Your e-mail will not be read or forwarded until then.

Thank you for your understanding.

Kind regards,

ing. Peter Boudewijns
CLICKEY solutions
***@***.***

CLICKEY solutions: +31-162-522052 / Fax: +31-162-519495
Lissenveld 47c, 4941 VL Raamsdonksveer, The Netherlands
http://www.clickey.eu

mluis1, 2021-12-13:
Thanks for the report.

In our internal repository we have already fixed this issue while re-working and implementing missing features for the ReJoinReq management. These fixes are still under validation that's why we did not made them public yet.

closed 2022-09-27 by mluis1