Lora-net/LoRaMac-node #1262: SX1276 don't go in TX mode

davide01frasson, 2022-02-07:
i'm coding a LoRa SX1276 transceiver from a Texas Instrument TMS320F28379D. Basically i copy-pasted the Arduino libraries changing the SPI protocol methods and everthing seemed fine, i was able to switch between SLEEP and STAND-BY mode, reading register and loading data into the Reg-Fifo and updating the Reg_Payload_Length, but when it comes to switch Reg_Op_Mode to TX mode it don't works. I've tried to put delays, changing power supply but i am not able to switch mode and to start transmission.

MedadRufus, 2022-02-07:
This repo does not deal with the Arduino libraries so you will not likely get any support here.

mluis1, 2022-03-21:
Please consider asking your question on a forum such https://forum.lora-developers.semtech.com/ you may get better support on the issues you are observing.

Another possibility is to use the radio drivers provided by this project which are known to be fully functional.

closed 2022-03-21 by mluis1