Lora-net/LoRaMac-node #1270: How to solve the Command Timeout of SX126x

blockchain4trust, 2022-02-21:
After running for about 2 days, the SX126x cannot receive under Class C!

We obtained the status word of SX1262 through the "probe interface" as
0x26=b00100110
bit6:4 010 0x2=STBY_RC
bit3:1 011 0x3=Command timeout

The manual for Command timeout explains it as follows. How to fix this bug Do you have any good suggestions?

[image]

MarekNovakACRIOS, 2022-03-09:
I encountered the same issue - my application is street light controller using class C.
This is related to this issue - https://github.com/Lora-net/LoRaMac-node/issues/1048 
Try to implement a workaround as suggested there - personally, I use LoRaMacIsBusy() function to check for the stack to be busy for too long (>30s?) and if this happens, then I force the stack to be IDLE:
```
void LoRaMacForceIdle(void)
{
    MacCtx.MacState = LORAMAC_IDLE;
}
```
By calling something like this.
It IS UGLY, but it saves the day for me - as I described in #1048 , I think this is somehow hardware related.

blockchain4trust, 2022-03-10:
[image]

Method 1: Read -> Confirm -> Fail to retransmit, execute 10 times. The bug still broke out.
Prove that the bug cannot be fixed by "confirmation".

[image]

Method 2 DisableIRQ before the SPI sends the command, and then EnableIRQ after the command ends, the error still occurs.
Prove that the bug is not caused by interrupted SPI communication.

Method 3 is OK. After nearly 10 days of testing, the bug has been basically solved---SX1262 has a Command Timeout exception during operation.

As shown in the figure below, the first two methods are invalid, the third method---add cron daemon, query and repair SX1262 exception every 30 seconds, success!

We write this event into EEPROM, read the log through the serial port, and find that the bug will appear once in about 3 days

vtjballeng, 2022-03-10:
@blockchain4trust , does network traffic impact your failure timeline?

blockchain4trust, 2022-03-11:
> @blockchain4trust , does network traffic impact your failure timeline?

This bug is not affected by network traffic. The measurement is performed at a fixed rate, with 160 Class C nodes sending a LoRa packet every 30 minutes.

mluis1, 2022-11-30:
This issue looks to be related to how the SX126x radio works.

We would recommend to discuss this problem under the [Semtech developer portal forum](https://forum.lora-developers.semtech.com/) and reference this issue as starting point. We believe that one will get better support by doing so.

If you don't mind we close the issue.

closed 2022-11-30 by mluis1