Lora-net/LoRaMac-node #1275: Out of bounds multicast group ID can cause memory corruption

gregbreen, 2022-03-01:
In handling several of the multicast package commands, the code uses the group ID directly from the incoming payload, and indexes into `McSessionData` array without range checking the ID value. For example:

```
            case REMOTE_MCAST_SETUP_MC_GROUP_SETUP_REQ:
            {
                uint8_t id = mcpsIndication->Buffer[cmdIndex++];
                McSessionData[id].McGroupData.IdHeader.Value = id;

                McSessionData[id].McGroupData.McAddr =  ( mcpsIndication->Buffer[cmdIndex++] << 0  ) & 0x000000FF;
                McSessionData[id].McGroupData.McAddr += ( mcpsIndication->Buffer[cmdIndex++] << 8  ) & 0x0000FF00;
                McSessionData[id].McGroupData.McAddr += ( mcpsIndication->Buffer[cmdIndex++] << 16 ) & 0x00FF0000;
                McSessionData[id].McGroupData.McAddr += ( mcpsIndication->Buffer[cmdIndex++] << 24 ) & 0xFF000000;

```

This is fine if `LORAMAC_MAX_MC_CTX` is 4. (Although in the above example, `id` could be as much as 255.) However if `LORAMAC_MAX_MC_CTX` is changed to a lower value (such as 1 in the STM32CubeWL firmware), this can be more problematic. The out of bounds array write occurs before an **IDerror** is determined.

If you agree, then REMOTE_MCAST_SETUP_MC_GROUP_CLASS_C_SESSION_REQ and REMOTE_MCAST_SETUP_MC_GROUP_CLASS_B_SESSION_REQ have a similar vulnerability. They are both ANDed with 0x03, but that has the assumption that LORAMAC_MAX_MC_CTX is 4.

MarekNovakACRIOS, 2022-03-09:
I have met a similar problem here - https://github.com/Lora-net/LoRaMac-node/pull/1277

mluis1, 2022-03-21:
Thanks for reporting this issue.
We will try to fix it for next release. Or maybe you could propose a PR to fix it.

mluis1, 2022-06-21:
Thanks for the feedback.

We will push the patch as soon as possible.

closed 2022-07-19 by mluis1