Lora-net/LoRaMac-node #1276: deveui / appeui all zero after updating se-identity.h

dudmuck, 2022-03-04:
i was trying 4.6.0 release on NucleoL073RZ, periodic-uplink-lpp with SOFT_SE.
After I edited se-identity.h with a devEui & joinEui and set STATIC_DEVICE_EUI to 1, then it was printing out on serial terminal zero values for those EUIs.
Then I randomly thought to try resetting NVM thru the serial terminal, and then after i reset the board, i could see the EUIs from the se-identity.h
If this is correct procedure, it should be documented?

mluis1, 2022-03-21:
Yes, it is the right procedure.

Every time one modifies a global parameter going to be stored on the NVM. This one should be reset in order to apply these new settings.

Could you propose a PR to improve the documentation?

mluis1, 2022-10-20:
Can we close this issue?

mluis1, 2022-11-24:
No answer if you don't mind we close the issue