Lora-net/LoRaMac-node #1284: Possibly wrong status returned by SwitchClass

janakj, 2022-03-14:
The function SwitchClass performs the following side-effect when the caller attempts to switch to class A when the device is in class A already (i.e., no change):

```c
    LoRaMacStatus_t status = LORAMAC_STATUS_PARAMETER_INVALID;

    switch( Nvm.MacGroup2.DeviceClass )
    {
        case CLASS_A:
        {
            if( deviceClass == CLASS_A )
            {
                // Revert back RxC parameters
                Nvm.MacGroup2.MacParams.RxCChannel = Nvm.MacGroup2.MacParams.Rx2Channel;
            }
    [...]
```
However, in this particular case the status value is not updated and the function returns LORAMAC_STATUS_PARAMETER_INVALID while performing the side-effect.

Shouldn't SwitchClass be updated to return LORAMAC_STATUS_OK if the side-effect (reverting RxC params) is performed?

MarekNovakACRIOS, 2022-03-14:
I have noticed that when evaluting class B - if the function has some side-effect, even when no actual change happened, it is error prone to return an error and it can easily mislead the user of the API. I definitely agree with @janakj .

mluis1, 2022-03-21:
Thanks for reporting this issue.

We will try to fix it for next release.