Lora-net/LoRaMac-node #1285: Promoting LORAMAC_VERSION and LORAMAC_FALLBACK_VERSION to the header?

janakj, 2022-03-15:
LoRaMac.c defines two macros potentially of interest to the application using the library: LORAMAC_VERSION and LORAMAC_FALLBACK_VERSION. Would you accept a PR moving the two macros to the header file LoRaMac.h so that they could be accessed from the application?

mluis1, 2022-03-16:
Could you please provide more context on why you think this proposition would be helpful?

Nowadays the MAC layer provides an API reporting which version is currently active. Isn't it enough?

Using `LoRaMacMibGetRequestConfirm` API one can read which LoRaWAN and Regional Parameters version is currently active.

https://github.com/Lora-net/LoRaMac-node/blob/bc9a93e1cbbea9e27b6af35dd542a96fc67f5699/src/mac/LoRaMac.c#L4084-L4089

janakj, 2022-03-17:
In my application, I would like to be able to obtain the most recent protocol version supported by the LoRaMac-node library, i.e., the value of the LORAMAC_VERSION macro. I wanted to use that value for automatic backend provisioning, i.e., to automatically add and configure the device in the TTN backend.

I tried to use the MIB API that you are suggesting for this purpose, however, the value returned by MIB_LORAWAN_VERSION is not stable and depends on whether the device is Joined, the MAC version supported by the LNS, or the MAC version configured for the device in the LNS. In case of TTN v3, you get either LORAMAC_VERSION or LORAMAC_FALLBACK_VERSION depending on the MAC version configured for the device in the TTN console. That's because of the following code in LoRaMac.c:

```c
// Is Networkserver's LoRaWAN Version before 1.1.0 ?
if( macMsgJoinAccept.DLSettings.Bits.OptNeg == 0 )
{
    Nvm.MacGroup2.Version.Value = LORAMAC_FALLBACK_VERSION;
}
else
{
    Nvm.MacGroup2.Version.Value = LORAMAC_VERSION;
} 
```

As far as I can tell, there is currently no way for an application to programmatically find out the MAC version supported by LoRaMac-node. Moving LORAMAC_VERSION to the header is a harmless change that would give applications an opportunity to obtain the protocol version supported by the library programmatically, regardless of the node's Join state.

mluis1, 2022-03-21:
Thanks for the context.

Instead of making `LORAMAC_VERSION` and `LORAMAC_FALLBACK_VERSION` visible through an header file I would prefer to have an API call able to return the currently supported versions (`LoRaMacMibGetRequestConfirm`).

We will try to provide an update as soon as we can.

janakj, 2022-03-21:
Hi Miguel,

Thank you for the feedback. Having an API makes sense for values that can dynamically change at runtime. MIB_LORAWAN_VERSION is a good example since the returned value depends on the MAC version negotiated with the network server.

For values statically defined via preprocessor macros such as LORAMAC_VERSION and LORAMAC_FALLBACK_VERSION it might be better to just export the macro to the application. That would be not only much simpler (no need to define new API), it would also make it possible for the application to access the value at compile time (not possible with an API). The values defined through LORAMAC_VERSION and LORAMAC_FALLBACK can only change during LoRaMac-node version updates.

Also, since LORAMAC_VERSION is surrounded by an `#ifndef` guard, I am assuming the intention was to allow the application to override the value. Thus, the application can set the value, but it cannot read the default value. That seems strange to me.

Anyway, all I am really suggesting is something like this:
```
diff --git a/src/mac/LoRaMac.c b/src/mac/LoRaMac.c
index 0c823882..e0d42867 100644
--- a/src/mac/LoRaMac.c
+++ b/src/mac/LoRaMac.c
@@ -48,18 +48,6 @@

 #include "LoRaMac.h"

-#ifndef LORAMAC_VERSION
-/*!
- * LoRaWAN version definition.
- */
-#define LORAMAC_VERSION                             0x01010100
-#endif
-
-/*!
- * LoRaWAN fallback version definition.
- */
-#define LORAMAC_FALLBACK_VERSION                    0x01000400
-
 /*!
  * Maximum PHY layer payload size
  */
diff --git a/src/mac/LoRaMac.h b/src/mac/LoRaMac.h
index c25a297d..046aa783 100644
--- a/src/mac/LoRaMac.h
+++ b/src/mac/LoRaMac.h
@@ -83,6 +83,18 @@ extern "C"
 #include "secure-element-nvm.h"
 #include "LoRaMacClassBNvm.h"

+#ifndef LORAMAC_VERSION^M
+/*!^M
+ * LoRaWAN version definition.^M
+ */^M
+#define LORAMAC_VERSION                             0x01010100^M
+#endif^M
+^M
+/*!^M
+ * LoRaWAN fallback version definition.^M
+ */^M
+#define LORAMAC_FALLBACK_VERSION                    0x01000400^M
+^M
 /*!
  * Maximum number of times the MAC layer tries to get an acknowledge.
  */
```

mluis1, 2022-03-22:
Sounds fair.

Could you please create a PR with your proposed changes?

Concerning the `#ifndef` I guess that we just have forgot to remove it at some point.

PR #1291 (2022-03-25): Move LORAMAC[_FALLBACK]?_VERSION macros to the header file
This trivial change makes the two version values accessible to the application using LoRaMac-node. Being able to obtain the most recent supported MAC version is useful in case the application needs to generate data, e.g., for backend (TTN) provisioning.

Fixes #1285
closed 2022-03-25 by mluis1