Lora-net/LoRaMac-node #1315: dutycycle bug in currentrelase v4.6.0

iot-lorawan, 2022-05-12:
Hi,
     I found  a dutycycle bug (dutycycle work failded in eu868 )in current relase in v4.6.0.

 #### Problem

 * actual behavior
 ----------the device can uplink up to 14 packets@DR0 with 10s period, and then continue uplink @DR0 SF12 with ~128s period when hit the dutycycle limit.( the total air time exceed 36s  in the first hour)

 * steps to reproduce
 ----------set joinmode with ABP and set APP_TX_DUTYCYCLE to 10s . 
THEN change eu868 to only send in 868.1 with blow code in regioneu868.c
 case INIT_TYPE_DEFAULTS:
        {
            if( ( params->NvmGroup1 == NULL ) || ( params->NvmGroup2 == NULL ) )
            {
                return;
            }

            RegionNvmGroup1 = (RegionNvmDataGroup1_t*) params->NvmGroup1;
            RegionNvmGroup2 = (RegionNvmDataGroup2_t*) params->NvmGroup2;
            RegionBands = (Band_t*) params->Bands;

            // Default bands
            memcpy1( ( uint8_t* )RegionBands, ( uint8_t* )bands, sizeof( Band_t ) * EU868_MAX_NB_BANDS );

            // Default channels
            RegionNvmGroup2->Channels[0] = ( ChannelParams_t ) EU868_LC1;
            //RegionNvmGroup2->Channels[1] = ( ChannelParams_t ) EU868_LC2;
            //RegionNvmGroup2->Channels[2] = ( ChannelParams_t ) EU868_LC3;

            // Default ChannelsMask
            RegionNvmGroup2->ChannelsDefaultMask[0] = LC( 1 );// + LC( 2 ) + LC( 3 );

            // Update the channels mask
            RegionCommonChanMaskCopy( RegionNvmGroup2->ChannelsMask, RegionNvmGroup2->ChannelsDefaultMask, CHANNELS_MASK_SIZE );
            break;
        }

djaeckle, 2022-06-03:
Hi @iot-lorawan ,

thanks for the report. Could you please provide more information about your setup? E.g.:

1. Which payload size are you transmitting?
2. Is ADR on or off?

iot-lorawan, 2022-06-06:
Hi @djaeckle,
    Which payload size are you transmitting?---------we use the default demoapplication with https://github.com/Lora-net/LoRaMac-node/blob/master/src/apps/LoRaMac/periodic-uplink-lpp/NucleoL073/main.c,  the payload size is 7 bytes
    Is ADR on or off?---------the device'adr is default on,but we turn the ns'adr setting to off  in networkserver

Best Regards,

closed 2022-11-24 by mluis1