Lora-net/LoRaMac-node #1318: In LoRaWAN 1.1, LastDownFCnt is out of sync

adriansmares, 2022-05-17:
Hello,

While working on the LoRaWAN 1.1 FOpts+FCntDown errata for The Things Stack (https://github.com/TheThingsNetwork/lorawan-stack/pull/5460), I've encountered the following bug - it seems that the last confirmed downlink frame counter used while computing the MIC of the uplink frame that confirms the downlink (so the next uplink after this confirmed downlink) is off by one.

I've tested this by adding the following debug statement:
```c
        printf("isAck true - LastDownFCnt: %d; AFCntDown: %d; NFCntDown: %d; FCntUp: %d", CryptoNvm->LastDownFCnt, CryptoNvm->FCntList.AFCntDown, CryptoNvm->FCntList.NFCntDown, CryptoNvm->FCntList.FCntUp);
```

At the following location (inside the `if`) in the 4.6.0 release:
https://github.com/Lora-net/LoRaMac-node/blob/e182832f8b79188c63b0737c184fb345932d50a9/src/mac/LoRaMacCrypto.c#L432-L438

I then follow the logs in the end device and it seems that the `LastDownFCnt` is never up to date:
```
isAck true - LastDownFCnt: -1; AFCntDown: 1; NFCntDown: 1; FCntUp: 5
isAck true - LastDownFCnt: 1; AFCntDown: 2; NFCntDown: 1; FCntUp: 7
isAck true - LastDownFCnt: 4; AFCntDown: 5; NFCntDown: 1; FCntUp: 13
```

It seems to be off by one from the relevant (either application or network) downlink frame counter.

I'm not entirely familiar with the LoRaMac-node codebase, but from what I can follow this _last_ frame counter is updated in 
1. `GetLastFcntDown`
https://github.com/Lora-net/LoRaMac-node/blob/e182832f8b79188c63b0737c184fb345932d50a9/src/mac/LoRaMacCrypto.c#L675
2. Which is called by `CheckFCntDown`
https://github.com/Lora-net/LoRaMac-node/blob/e182832f8b79188c63b0737c184fb345932d50a9/src/mac/LoRaMacCrypto.c#L729
3. Which is called by `LoRaMacCryptoUnsecureMessage`
https://github.com/Lora-net/LoRaMac-node/blob/e182832f8b79188c63b0737c184fb345932d50a9/src/mac/LoRaMacCrypto.c#L875
4. Which is called by the downlink handling code
https://github.com/Lora-net/LoRaMac-node/blob/e182832f8b79188c63b0737c184fb345932d50a9/src/mac/LoRaMac.c#L1175

There is also the stack trace:
1. (same as above)
2. Which is called by `LoRaMacCryptoGetFCntDown`
https://github.com/Lora-net/LoRaMac-node/blob/e182832f8b79188c63b0737c184fb345932d50a9/src/mac/LoRaMacCrypto.c#L875
3. Which is called by `GetFCntDown`
https://github.com/Lora-net/LoRaMac-node/blob/e182832f8b79188c63b0737c184fb345932d50a9/src/mac/LoRaMac.c#L1895
4. Which is called by the downlink handling code
https://github.com/Lora-net/LoRaMac-node/blob/e182832f8b79188c63b0737c184fb345932d50a9/src/mac/LoRaMac.c#L1157

I think that the bug is that since the `LastDownFCnt` is updated on `Get`, instead of `Update`, it basically uses not the last downlink frame counter, but rather the one before, as the flow is as follows:

1. Check the frame counter using `GetFCntDown` - this will update `LastFCntDown` to the frame counter _before_ processing the current downlink frame
2. Decrypt the frame using `LoRaMacCryptoUnsecureMessage` - this again will update the frame counter to the frame counter _before_ processing the current downlink frame (since we didn't update the NVM, and we're only doing `Get` operations)
3. Actually save the frame counter that needs to be confirmed as part of `LoRaMacCryptoUnsecureMessage`
https://github.com/Lora-net/LoRaMac-node/blob/e182832f8b79188c63b0737c184fb345932d50a9/src/mac/LoRaMacCrypto.c#L1462

It seems to me that saving the `LastFCntDown` on `Get` instead of `Update` was a deliberate choice, but in the current flow this seems wrong. Am I missing something ?

mluis1, 2022-06-15:
First I would like to thank you for the nice issue description.

We need some time to investigate the observed behavior. We will try to do it as soon as possible.

Could you confirm that the way to reproduce the issue is by scheduling Confirmed downlinks under the AS server?

The last time we checked 1.1.0 behavior we used ChirpStack network server and I don't remember to have seen this issue.

adriansmares, 2022-06-15:
> Could you confirm that the way to reproduce the issue is by scheduling Confirmed downlinks under the AS server?

That is indeed how I did it. 

I would argue that the smoking gun here is that the `LastFCntDown` used to compute the MIC is never really the frame counter of the frame that is being confirmed. Even the `-1` value is bogus, and it will be used if the first downlink is a confirmed downlink.

closed 2022-07-19 by mluis1