Lora-net/LoRaMac-node #1359: LmHandler - Inconsistent Handling of undefined Callbacks

dmeehan1968, 2022-10-10:
Few of the LmHandler callbacks are guarded against NULL (you can call LmHandlerInit with NULL values for any of the callbacks without it returning an error, and the callbacks are used without validation, which can cause a fault.

Some of the underlying LoRaMac callbacks (GetBatteryLevel etc) are partially guarded but not entirely consistently.

mluis1, 2022-10-11:
Thanks for the proposition.

Would you be wiling to create a pull-request providing those NULL guards.

dmeehan1968, 2022-10-11:
I think I need a steer on whether `LmHandlerInit` should validate that they are non-null, or guard the call in the event of `NULL`.  I think this depends on whether some of the callbacks are considered mandatory or not.  If you can clarify then I can look at a PR.

mluis1, 2022-10-11:
Searching which callbacks require a guard and which do not I found that at least one requires a change to `LoRaMac.c` file.
https://github.com/Lora-net/LoRaMac-node/blob/e851b079c82ba1bcf3f4d291ab69a571b0bf458a/src/mac/LoRaMac.c#L5272

I have also noticed that we could optimize the following as it appears several times.
https://github.com/Lora-net/LoRaMac-node/blob/e851b079c82ba1bcf3f4d291ab69a571b0bf458a/src/mac/LoRaMac.c#L736

I will propose the required changes as soon as possible. If no public API changes I will target v4.7.0 milestone. In case there are changes we will target v5.0.0 milestone.

mluis1, 2022-10-11:
Please find below the patch which should apply your proposition.

I think that I have covered all callback calls. Could you please check on your side as well?

```patch
diff --git a/src/apps/LoRaMac/common/LmHandler/LmHandler.c b/src/apps/LoRaMac/common/LmHandler/LmHandler.c
index 3ebdcd54..db8875f3 100644
--- a/src/apps/LoRaMac/common/LmHandler/LmHandler.c
+++ b/src/apps/LoRaMac/common/LmHandler/LmHandler.c
@@ -186,6 +186,32 @@ static void MlmeConfirm( MlmeConfirm_t *mlmeConfirm );
  */
 static void MlmeIndication( MlmeIndication_t *mlmeIndication );
 
+/*!
+ * Calls the OnClassChange callback to indicate the new active device class
+ *
+ * \param [in] deviceClass Active device class
+ */
+static void OnClassChangeNotify( DeviceClass_t deviceClass )
+{
+    if( LmHandlerCallbacks->OnClassChange != NULL )
+    {
+        LmHandlerCallbacks->OnClassChange( deviceClass );
+    }
+}
+
+/*!
+ * Calls the OnBeaconStatusChange callback to indicate current beacon status
+ *
+ * \param [in] params Current beacon parameters
+ */
+static void OnBeaconStatusChangeNotify( LoRaMacHandlerBeaconParams_t *params )
+{
+    if( LmHandlerCallbacks->OnBeaconStatusChange != NULL )
+    {
+        LmHandlerCallbacks->OnBeaconStatusChange( params );
+    }
+}
+
 /*!
  * Starts the beacon search
  *
@@ -252,7 +278,7 @@ LmHandlerErrorStatus_t LmHandlerInit( LmHandlerCallbacks_t *handlerCallbacks,
     nbNvmData = NvmDataMgmtRestore( );
 
     // Try to restore from NVM and query the mac if possible.
-    if( nbNvmData > 0 )
+    if( ( LmHandlerCallbacks->OnNvmDataChange != NULL ) && ( nbNvmData > 0 ) )
     {
         LmHandlerCallbacks->OnNvmDataChange( LORAMAC_HANDLER_NVM_RESTORE, nbNvmData );
     }
@@ -278,8 +304,12 @@ LmHandlerErrorStatus_t LmHandlerInit( LmHandlerCallbacks_t *handlerCallbacks,
         LoRaMacMibSetRequestConfirm( &mibReq );
 
 #if( STATIC_DEVICE_ADDRESS != 1 )
+        // TODO: Remove STATIC_DEVICE_ADDRESS = 0. Up to the application to decide what to do
         // Random seed initialization
-        srand1( LmHandlerCallbacks->GetRandomSeed( ) );
+        if( LmHandlerCallbacks->GetRandomSeed != NULL )
+        {
+            srand1( LmHandlerCallbacks->GetRandomSeed( ) );
+        }
         // Choose a random device address
         CommissioningParams.DevAddr = randr( 0, 0x01FFFFFF );
 #endif
@@ -320,7 +350,10 @@ LmHandlerErrorStatus_t LmHandlerInit( LmHandlerCallbacks_t *handlerCallbacks,
     {
         if( mibReq.Param.NetworkActivation == ACTIVATION_TYPE_NONE )
         {
-            LmHandlerCallbacks->OnNetworkParametersChange( &CommissioningParams );
[... 188 lines cut ...]
-            LmHandlerCallbacks->OnBeaconStatusChange( &BeaconParams );
+            OnClassChangeNotify( CLASS_A );
+            OnBeaconStatusChangeNotify( &BeaconParams );
 
             LmHandlerDeviceTimeReq( );
         }
@@ -880,14 +951,14 @@ static void MlmeIndication( MlmeIndication_t *mlmeIndication )
             BeaconParams.State = LORAMAC_HANDLER_BEACON_RX;
             BeaconParams.Info = mlmeIndication->BeaconInfo;
 
-            LmHandlerCallbacks->OnBeaconStatusChange( &BeaconParams );
+            OnBeaconStatusChangeNotify( &BeaconParams );
         }
         else
         {
             BeaconParams.State = LORAMAC_HANDLER_BEACON_NRX;
             BeaconParams.Info = mlmeIndication->BeaconInfo;
 
-            LmHandlerCallbacks->OnBeaconStatusChange( &BeaconParams );
+            OnBeaconStatusChangeNotify( &BeaconParams );
         }
         break;
     }

```

dmeehan1968, 2022-10-11:
I didn't apply it, as I doubt I could effectively test it.   I read through it and made the following observations:

1. `LoRaMac.c`: `IsRequestPending` - You've changed the evaluation order of the callback and `MacDone` which might be important.
2. `LoRaMacClassB.c`: Using the same helper name as in LoRaMac.c, even if its a `static` might be confusing as its accessing a different global.  Would be inclined to rename this one to include the ClassB designation so its clearer, but you may have a view that its consistent with the rest of that module.

If any of the callbacks are critical to functionality (like MacProcessNotify) then these should really be validated in LoRaMacInitialization/LmHandlerInit so someone can't use the stack erroneously (as I did!).

mluis1, 2022-10-13:
Thanks fr the feedback.

1. I will revert to original sequence. However I think the change is not harmful,
2. I will rename the helper function

I believe that all mandatory callbacks have a check.

`MacProcessNotify` is not a mandatory callback. The goal of this notification is to inform the application layer that there is some MAC layer processing pending. In case calls to `LoRaMacProcess` are done everything should work.
We have initially added `MacProcessNotify` in order to avoid going to a MCU Sleep mode before all pending MAC processing is finished. In provided examples if one removes the low power mode management there is no need for `MacProcessNotify` as we have the insurance that `LoRaMacProcess` function will always be called.

closed 2022-10-13 by mluis1