Lora-net/LoRaMac-node #1403: Join procedure completion for Class C

wm-eisos, 2023-01-10:
According to the Lorawan 1.0.4 specification there is a specific procedure for class c and OTAA which doesn't seem to be implemented in the stack
[image]

mluis1, 2023-01-10:
This is implemented and in addition it is verified by the LoRaWAN certification process.

Even if the application layer tries to perform a first uplink of type Unconfirmed the MAC layer will force it to be of Confirmed type. Please refer to:
https://github.com/Lora-net/LoRaMac-node/blob/672b37deec7589b2e7346d3acb6a0e8ba79fa9fe/src/mac/LoRaMac.c#L5498-L5506

Please note that I just noticed a typo in the comment. One should read "Apply confirmed uplinks,"" instead of "Apply confirmed downlinks,"

mluis1, 2023-01-10:
It is up to the application layer to ask the MAC layer to transmit uplinks.

Yes, this confirmed uplink has been introduced in 1.0.4 in order to solve issues observed on previous version of the specification.

In 1.0.3 specification this behavior is not specified then all possible combinations are valid which means all kind of weird behaviors can happen if not carefully handled.

If you don't mind I close the issue.

closed 2023-01-10 by mluis1