Lora-net/LoRaMac-node #1408: Floating point usage in radio.c

kritzikratzi, 2023-02-11:
copy and pasting my bug report from https://github.com/HelTecAutomation/CubeCell-Arduino/issues/261

radio.c uses floating point numbers (doubles, to be precise) to calculate the air time (maybe there are other places too), however it would be quite trivial to rewrite this computation to fixed point maths.

the goal would be to get rid of the generated code for float/double support for embedded systems without floating point support:

8035: 0000f7bd 1464 FUNC GLOBAL HIDDEN 2 __aeabi_ddiv
8141: 0000f0d1 1772 FUNC GLOBAL HIDDEN 2 __aeabi_dadd
8045: 0000ffb1 1252 FUNC GLOBAL HIDDEN 2 __aeabi_dmul
8097: 0000ecc9 840 FUNC GLOBAL HIDDEN 2 __aeabi_fsub
8279: 0000ea99 560 FUNC GLOBAL HIDDEN 2 __aeabi_fdiv

in total this accounts for 5.8kB of flash.

danak6jq, 2023-02-11:
I haven't looked yet, but how much precision do these calculations require? 32-bits?

kritzikratzi, 2023-02-12:
good question. 

frequency is among the inputs, and is in the high mhz range. 

time is the output, and is in the milliseconds range. 

that's a larger value span than 32 bit integer can do, so a bit of scaling here and there will be required (maybe it's as simple as dividing the frequencies by 1000 first, and then the result outmatically comes out as milliseconds integers and the problem magically goes away, but i don't have the code in front of me and don't know).

danak6jq, 2023-02-12:
I note that ST re-wrote RadioTimeOnAir() to use integer math. Probably a good example here. I'm using the STM32WL firmware v1.3.

mluis1, 2023-02-13:
I don't know which version of this project you are using.
However, since October 2020 (v4.4.5) the time on air functions as well as functions handling frequencies for all supported radios do use integer math based computations.
Please refer to CHANGELOG.md file [[4.4.5] - 2020-10-14](https://github.com/Lora-net/LoRaMac-node/blob/master/CHANGELOG.md#445---2020-10-14)

### Changed
 
 - Refactored function RegionXXInitDefaults. Fixes an issue on US based bands where the channels mask was not being reset.
 - Changed hard coded JoinAccept max payload size (33) by LORAMAC_JOIN_ACCEPT_FRAME_MAX_SIZE definition.
 - Moved radio operating mode management to specific board implementation
 - Changed radio IsChannelFree API in order to provide reception bandwidth
 - AS923 default downlink dwell time setting changed to disabled
 - Back port 1.0.4 region implementation improvements
 - **Changed RegionCommonComputeSymbolTimeLoRa, RegionCommonComputeSymbolTimeFsk and RegionCommonComputeRxWindowParameters API implementations to use integer divisions instead of double division.**
 - Changed DeriveSessionKey10x and DeriveSessionKey11x functions API (Removed pointer based variable usage)

At some point in time we have tried to remove all `double`/`float` usage. 
The only places where `double`/`float` are still in use are `gps.c/h`, functions handling temperature `rtc-board.c/h` and `CayenneLpp.c/h`.

Please take a look at this project [master](https://github.com/Lora-net/LoRaMac-node/tree/master) branch radio drivers implementation. You may also take a look at [v5.0.0-branch](https://github.com/Lora-net/LoRaMac-node/tree/v5.0.0-branch) as it implements completely refactored radio drivers.

kritzikratzi, 2023-02-13:
thanks. i will pass on the information.

closed 2023-02-13 by kritzikratzi