Lora-net/LoRaMac-node #1527: Long range at sea : 10-20 km expected, 300m obtained !!

quentinorh, 2023-03-26:
Hi,

I'm trying to communicate in Lora with Adafruit Feather M0 in 433mhz.

The objective is to be able to transmit the GPS position of a beacon in the open sea, and to find it by boat, probably with a Yagi antenna.

I have two M0 433mhz cards, one with a 16.5cm wire antenna, the other with a yagi antenna (9 dBi) and a 5w power amplifier (see photo).

[image]

I have just done some tests, I can't manage to receive a package at more than 300m... whereas I would need 10-20km!

Am I too optimistic, or am I wrong in the setting / electronics?

danak6jq, 2023-03-26:
You're asking a radio question, not so much a LoRaWAN question.

I note that amplifier you're using (I believe I have the same one) is incomplete without (considerably) more low-pass filtering on the output. In fact, I'm not even sure what the performance of that amplifier at 433MHz is in terms of output matching. Have you verified that you're not over-driving it, and that you're switching it out of the antenna path on receive (this may explain the limited range right there)?

mluis1, 2023-03-27:
As mentioned by @danak6jq this question is not related to this project. This project is a software library implementing the LoRaWAN protocol stack.

If you do not mind I move this question to the [Discussions](https://github.com/Lora-net/LoRaMac-node/discussions) tab.

A better place to ask these kind of questions is the [LoRa Developers forum](https://forum.lora-developers.semtech.com/)