Lora-net/LoRaMac-node #1531: NVM Restore issue

AD-NCD, 2023-04-04:
when i reset my device, i never get data restored. I always get a store event instead, when i tracked this. I found that CRC isn't matched starting from secure element to Class b all of them aren't valid, Although when i checked sector element on nvm after reset i was able to read data from it and data were correct
Has any one used this?

satukbugratekin, 2023-04-04:
Would it be that you are trying to Join again after reset, even you already joined to the network?

AD-NCD, 2023-04-04:
> Would it be that you are trying to Join again after reset, even you already joined to the network?
it's not the case on my side, when i reset i call LmHandlerInit, on it there's a call to NvmDataMgmtRestore and Crc fails there.
It's before joining i believe if data is restored then joining will be a pass through API

satukbugratekin, 2023-04-04:
Since the actual join status is not checked by any mechanism in Lorawan but should be done in Application by sending some heartbeat, if this is implemented then you would never understand if the connection is lost, mine is already several uears ago and already running somehow 😀 

As I said I was thinking that Join is invalidating data on your case. But seems this is not the case for your side.

closed 2023-04-11 by AD-NCD