Lora-net/LoRaMac-node #1564: LR1110MB1XXS with STM32L476, sleep mode settings

sohaibqamar, 2023-09-16:
Hello everyone, 
I am using a custom board for a LoRaWAN-based solution and using this stack for the development. 
I'm using the below settings.
```
"APPLICATION":"LoRaMac",
"SUB_PROJECT":"periodic-uplink-lpp",
"LORAWAN_DEFAULT_CLASS":"CLASS_A",
"CLASSB_ENABLED":"ON",
"MODULATION":"LORA",
"BOARD":"NucleoL476",
"MBED_RADIO_SHIELD":"LR1110MB1XXS",
"SECURE_ELEMENT":"SOFT_SE"
```
I have included a GPS over UART in the project and used some GPIOs for LEDs and Buzzer. I am using AWS IoT Core for Uplink/Downlinks which is working perfectly fine. Periodic time for sending uplink is 1min.
My system is not going into sleep mode or any low-power mode using the default while(1) loop.
I am measuring the current using a power profiler and the system current is around 50-60mA, I am unable to go into any sleep mode.
Any help would be appreciated.

sohaibqamar, 2023-09-25:
My main function looks like this after I deleted the portion of interfacing the Accelerometer and GPS

```
int main( void )
{
    BoardInitMcu( );
    BoardInitPeriph( );
    TimerInit( &Led1Timer, OnLed1TimerEvent );
    TimerSetValue( &Led1Timer, 25 );
    TimerInit( &Led2Timer, OnLed2TimerEvent );
    TimerSetValue( &Led2Timer, 25 );
    TimerInit( &LedBeaconTimer, OnLedBeaconTimerEvent ); // What is this timer doing
    // TimerSetValue( &LedBeaconTimer, 5000 ); // original
    TimerSetValue( &LedBeaconTimer, 30000 );
    // Initialize transmission periodicity variable
    TxPeriodicity = APP_TX_DUTYCYCLE + randr( -APP_TX_DUTYCYCLE_RND, APP_TX_DUTYCYCLE_RND );
    const Version_t appVersion = { .Value = FIRMWARE_VERSION };
    const Version_t gitHubVersion = { .Value = GITHUB_VERSION };
    DisplayAppInfo( "periodic-uplink-lpp", 
                    &appVersion,
                    &gitHubVersion );    
    if ( LmHandlerInit( &LmHandlerCallbacks, &LmHandlerParams ) != LORAMAC_HANDLER_SUCCESS )
    {
        printf( "LoRaMac wasn't properly initialized\n" );
        // Fatal error, endless loop.
        while ( 1 )
        {
            printf( "* " );
        }
    }
    // Set system maximum tolerated rx error in milliseconds
    LmHandlerSetSystemMaxRxError( 20 );
    // The LoRa-Alliance Compliance protocol package should always be
    // initialized and activated.
    LmHandlerPackageRegister( PACKAGE_ID_COMPLIANCE, &LmhpComplianceParams );
    LmHandlerJoin( );
    StartTxProcess( LORAMAC_HANDLER_TX_ON_TIMER );    
    while( 1 )
    {
        // Processes the LoRaMac events
        LmHandlerProcess( );
        // Process application uplinks management
        UplinkProcess( );
        CRITICAL_SECTION_BEGIN( );
        if( IsMacProcessPending == 1 )
        {
            // Clear flag and prevent MCU to go into low power modes.
            IsMacProcessPending = 0;
        }
        else
        {
            // The MCU wakes up through events
            BoardLowPowerHandler( );
            //printf("Wake up.\n");
        }
        CRITICAL_SECTION_END( );
    }
}
```
In function `void BoardInitMcu( void )` i have set `UsbIsConnected = false;`

mluis1, 2023-09-26:
Is is hard to help you figure out why such high power consumption is observed.

There are multiple reasons to have such behaviors.
As you indicate that you have your own custom board it is even harder to help you as we do not know how your board has been designed.

The only thing we can do is to provide some clues on what you should look at.
We suppose that `BoardLowPowerHandler` function gets called. This function execution is where the MCU enters the sleep modes.

Things to verify related to your custom board:
- Are there any LED that is ON?
- Check that MCU pins are not floating or are set to the right level when in sleep
- Check that there is no debug probe connected to your custom board
- Are external peripheral component set in sleep mode as well (temperature sensors, GPS receivers, etc)
- etc

Debugging why one observes a higher power consumption than expected is a hard task and very time consuming.

In the future it would be nice if you could post this kind of questions on the project [Discussions](https://github.com/Lora-net/LoRaMac-node/discussions) tab. It is a better place to engage discussions and then we can agree if it is an issue or not.
If you don't mind I move this issue to the [Discussions](https://github.com/Lora-net/LoRaMac-node/discussions) tab.

closed 2023-09-26 by mluis1