Lora-net/LoRaMac-node #1587: TimerInit called multiple times on the same TimerEvent_t struct

ndiddy99, 2023-11-28:
Hi,

On a Class B device, calling [LmHandlerJoin](https://github.com/Lora-net/LoRaMac-node/blob/6384af8f4c4a57016c83f3d34430f0f68855e695/src/apps/LoRaMac/common/LmHandler/LmHandler.c#L467) will call [LmHandlerJoinRequest](https://github.com/Lora-net/LoRaMac-node/blob/6384af8f4c4a57016c83f3d34430f0f68855e695/src/apps/LoRaMac/common/LmHandler/LmHandler.c#L439), which calls [LoRaMacMlmeRequest](https://github.com/Lora-net/LoRaMac-node/blob/6384af8f4c4a57016c83f3d34430f0f68855e695/src/mac/LoRaMac.c#L5315), which calls [ResetMacParameters](https://github.com/Lora-net/LoRaMac-node/blob/6384af8f4c4a57016c83f3d34430f0f68855e695/src/mac/LoRaMac.c#L3143), which calls [LoRaMacClassBInit](https://github.com/Lora-net/LoRaMac-node/blob/6384af8f4c4a57016c83f3d34430f0f68855e695/src/mac/LoRaMacClassB.c#L619).

The issue is that the 3 timers in that function will get TimerInit called on them multiple times if there's multiple join requests. TimerInit will set a TimerEvent_t's Next pointer to NULL, which can mess up the timer linked list. Our internal fix was to check if the timers were initialized prior to calling TimerInit:

```
void LoRaMacClassBInit( LoRaMacClassBParams_t *classBParams, LoRaMacClassBCallback_t *callbacks, LoRaMacClassBNvmData_t* nvm )
{
#ifdef LORAMAC_CLASSB_ENABLED
    static bool timersInitialized = false;

    // Assign non-volatile context
    if( nvm == NULL )
    {
        return;
    }
    ClassBNvm = nvm;

    // Store callbacks
    Ctx.LoRaMacClassBCallbacks = *callbacks;

    // Store parameter pointers
    Ctx.LoRaMacClassBParams = *classBParams;

    // Initialize timers
    if ( !timersInitialized )
    {
        timersInitialized = true;
        TimerInit( &Ctx.BeaconTimer, LoRaMacClassBBeaconTimerEvent );
        TimerInit( &Ctx.PingSlotTimer, LoRaMacClassBPingSlotTimerEvent );
        TimerInit( &Ctx.MulticastSlotTimer, LoRaMacClassBMulticastSlotTimerEvent );
    }

    InitClassB( );
#endif // LORAMAC_CLASSB_ENABLED
}
```

I can submit a PR if you think that's an acceptable fix.

ndiddy99, 2023-12-01:
This might be a better fix (not sure if stopping the timers is necessary but it makes me feel better)

```
void LoRaMacClassBInit( LoRaMacClassBParams_t *classBParams, LoRaMacClassBCallback_t *callbacks, LoRaMacClassBNvmData_t* nvm )
{
#ifdef LORAMAC_CLASSB_ENABLED
    static bool timersInitialized = false;

    // Assign non-volatile context
    if( nvm == NULL )
    {
        return;
    }
    ClassBNvm = nvm;

    // Store callbacks
    Ctx.LoRaMacClassBCallbacks = *callbacks;

    // Store parameter pointers
    Ctx.LoRaMacClassBParams = *classBParams;

    // Initialize timers
    if ( !timersInitialized )
    {
        timersInitialized = true;
        TimerInit( &Ctx.BeaconTimer, LoRaMacClassBBeaconTimerEvent );
        TimerInit( &Ctx.PingSlotTimer, LoRaMacClassBPingSlotTimerEvent );
        TimerInit( &Ctx.MulticastSlotTimer, LoRaMacClassBMulticastSlotTimerEvent );
    }
    else
    {
        if (Ctx.BeaconTimer.IsStarted)        { TimerStop(&Ctx.BeaconTimer); }
        if (Ctx.PingSlotTimer.IsStarted)      { TimerStop(&Ctx.PingSlotTimer); }
        if (Ctx.MulticastSlotTimer.IsStarted) { TimerStop(&Ctx.MulticastSlotTimer); }
    }

    InitClassB( );
#endif // LORAMAC_CLASSB_ENABLED
}
```

ClaireJWyld, 2024-04-03:
I did it a different way and fixed TimerInit in timer.c, checking if the timer object already exists in the linked list, and if so, stopping it before initialising.

`void TimerInit( TimerEvent_t *obj, void ( *callback )( void *context ) )
{
    if (TimerExists(obj))
    {
        TimerStop(obj);
    }

    obj->Timestamp = 0;
    obj->ReloadValue = 0;
    obj->IsStarted = false;
    obj->IsNext2Expire = false;
    obj->Callback = callback;
    obj->Context = NULL;
    obj->Next = NULL;
}`