Lora-net/LoRaMac-node #1594: LmHandlerInit causing some current leakage till the first tx

AD-NCD, 2024-01-09:
Hello,
i was making power profiling and i found there is always a leakage before first Transmission "doesn't matter it's a join or anything else". 
After first transmission everything turns to be fine. I am using LmHandlerInit to init mac, i found that after setting public network param leakage occurs which seems to be writing on radio and radio doesn't go to sleep after it when i removed this part issue solved.
.
Here's the code line 328:
     
    mibReq.Type = MIB_PUBLIC_NETWORK;
    mibReq.Param.EnablePublicNetwork = LmHandlerParams->PublicNetworkEnable;
    LoRaMacMibSetRequestConfirm(&mibReq);
I think the solution is to sleep radio at the end of LmHandlerInit.
2 images attached for the power profiling. Power seems to be higher than usual because i am having some extra hw connected 

[image]
[image]

AD-NCD, 2024-01-10:
After tracing. In loramac.c file line 4686
        case MIB_PUBLIC_NETWORK:
        {
            Nvm.MacGroup2.PublicNetwork = mibSet->Param.EnablePublicNetwork;
            Radio.SetPublicNetwork( Nvm.MacGroup2.PublicNetwork );
            break;
        }
setting this command in radio wakes it up so we should be sleeping radio after this
Same on Restoring NVM line 3512
If i added Radio.sleep()
Everything works fine

lff5, 2024-06-13:
I have the same issue with Restoring from NVM. Radio.sleep() must be called afterwards.

AD-NCD, 2024-07-08:
Still not solved!