Lora-net/LoRaMac-node #1616: LA915 region not supported

anacforcelli, 2024-08-29:
I am working from Brazil and have been struggling for a while to make my company's custom board work on our LoRaWAN network (hosted by American Tower). 

Recently I found out that the AU915 localization, whilst similar to LA915 specification, does not work seamlessly and there is a lot of packet loss that may be due to different RX/TX window differences. 

I am attempting to change the parameters as of now and may report later today. If successful, a PR will be submitted. Otherwise, I will return for help. 

Thanks in advance!

danak6jq, 2024-08-30:
Is this a request to support the LA915 region not found in the LoRaWAN Regional Parameters specs?

anacforcelli, 2024-08-31:
yes, I'm trying to do this but somehow it does not compile correctly, maybe i'm doing something wrong with the configs...

danak6jq, 2024-08-31:
What build errors are you seeing? (@mluis1 might convert this to a discussion )