Lora-net/LoRaMac-node #1628: Improve porting procedure via CMake for projects which do not use LoRaMac-node as base project

ckrenslehner, 2024-12-11:
Hi! 

I integrated the current state of this repository in my firmware. I took a look at the porting guide, but the guide requires that the project is based on this repository as root so to say. 
If this is the case, it is rather easy to add a new board and one should be fine. 

In my case, however, I have an own cmake root project and just want to integrate this stack into my existing project. I achieved this by avoiding including the main CMakeLists.txt of this project and just picking what I need.

Here is the relevant snipped of my `CMakeLists.txt`:

```
  # I want to use some of the common components. So I made a library to which I can link to
  add_library(loramac-handler STATIC "")
  target_include_directories(
    loramac-handler
    PUBLIC LoRaMac-node/src/apps/LoRaMac/common
           LoRaMac-node/src/apps/LoRaMac/common/LmHandler
           LoRaMac-node/src/apps/LoRaMac/common/LmHandler/packages
           LoRaMac-node/src/system)
  target_sources(
    loramac-handler
    PRIVATE
      LoRaMac-node/src/apps/LoRaMac/common/NvmDataMgmt.c
      LoRaMac-node/src/apps/LoRaMac/common/LmHandler/packages/FragDecoder.c
      LoRaMac-node/src/apps/LoRaMac/common/LmHandler/packages/FragDecoder.h
      LoRaMac-node/src/apps/LoRaMac/common/LmHandler/packages/LmhPackage.h
      LoRaMac-node/src/apps/LoRaMac/common/LmHandler/packages/LmhpClockSync.c
      LoRaMac-node/src/apps/LoRaMac/common/LmHandler/packages/LmhpClockSync.h
      LoRaMac-node/src/apps/LoRaMac/common/LmHandler/packages/LmhpCompliance.c
      LoRaMac-node/src/apps/LoRaMac/common/LmHandler/packages/LmhpCompliance.h
      LoRaMac-node/src/apps/LoRaMac/common/LmHandler/packages/LmhpFragmentation.c
      LoRaMac-node/src/apps/LoRaMac/common/LmHandler/packages/LmhpFragmentation.h
      LoRaMac-node/src/apps/LoRaMac/common/LmHandler/packages/LmhpRemoteMcastSetup.c
      LoRaMac-node/src/apps/LoRaMac/common/LmHandler/packages/LmhpRemoteMcastSetup.h
      LoRaMac-node/src/apps/LoRaMac/common/LmHandler/LmHandler.c
      LoRaMac-node/src/apps/LoRaMac/common/LmHandler/LmHandler.h
      LoRaMac-node/src/apps/LoRaMac/common/LmHandler/LmHandlerTypes.h
      LoRaMac-node/src/boards/mcu/utilities.c
      LoRaMac-node/src/radio/sx126x/sx126x.c
      LoRaMac-node/src/radio/sx126x/radio.c)

  # Gather all the other libraries
  target_link_libraries(loramac-handler PUBLIC board)
  target_link_libraries(loramac-handler PUBLIC system)
  target_link_libraries(loramac-handler PUBLIC peripherals)
  target_link_libraries(loramac-handler PUBLIC radio)
  target_link_libraries(loramac-handler PUBLIC mac)

  # "loramac-board" is my own static library which includes the board support for my custom hardware
  set(BOARD "loramac-board")

  # Include the subdirs
  add_subdirectory(LoRaMac-node/src/boards)
  add_subdirectory(LoRaMac-node/src/system)

  set(SECURE_ELEMENT "SOFT_SE")
  add_subdirectory(LoRaMac-node/src/peripherals)

  set(RADIO "sx126x")
  add_subdirectory(LoRaMac-node/src/radio)

  set(REGION_EU868
      ON
      CACHE BOOL "Enable REGION_EU868")
  add_subdirectory(LoRaMac-node/src/mac)
```

Maybe the procedure can be improved or this snipped can be added to the porting guide. :-)

Greetings,
Christian

nfroggy, 2024-12-30:
You probably don't want to be using this project for anything new. Semtech has left important issues unaddressed for over a year at this point.

ckrenslehner, 2024-12-31:
Really? What would you recommend as an alternative?

nfroggy, 2024-12-31:
LoRa Basics Modem is Semtech's new reference stack that they're currently putting all their development effort into. Of course, you run the risk of Semtech stopping development and forcing users to port their code to another stack if they want updates like what happened with LoRaMac-node. If you're developing a commercial product that requires ongoing support, that's a problem. Your microcontroller vendor may also maintain their own LoRaWAN stack. We looked into ST's when we were starting development (it's a fork of LoRaMac-node), but when we evaluated it around a year and a half ago it didn't pass the LCTT compliance test suite.

ckrenslehner, 2025-01-01:
Thank you, I will take a look at these options. :-)